agherzan/meta-raspberrypi #8: Most of the initscripts in rcS.d are not being executed

zwerch, 2016-02-19:
# Summary

We found out that most of the initscripts in `/etc/rcS.d` are not executed. In fact only `S00psplash` is executed as far as we could find out. We discovered this because `S04udev` seemed not to be running.
We could not find the cause though, if we work around this by starting the scripts manually it works like a charm (but this doesn't feel right).

# Steps to reproduce

We wanted to build our Yocto-based firmware that already existed for another machine now for the `raspberrypi2`, so we added this layer and baked an image. We were on `dizzy` back then. Then we found this bug which did not exist on our Freescale iMX.6 based machine we used before.
We baked a few different images for debugging (`rpi-hwup-image`, `rpi-test-image`, `core-image-x11`) and experienced the bug on every one of them.
We updated the whole project to `jethro` afterwards, hoping that it would fix the bug. It did not.

# Expected result

The initscripts should all have been executed after boot.

# Actual result

We put some `echo`-logs in the different `S0X`-files and it showed that only `S00psplash` got executed.

# Logs / files etc.

## Build configuration 

```
Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Debian-8.0"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "raspberrypi2"
DISTRO            = "solarfox"
DISTRO_VERSION    = "2.0.1"
TUNE_FEATURES     = "arm armv7a vfp thumb neon callconvention-hard vfpv4 cortexa7"
TARGET_FPU        = "vfp-vfpv4-neon"
meta              
meta-yocto        = "jethro:5b12268f6e17574999f91628a60e21711cf62ee4"
meta-browser      = "jethro:e114d625d4bd23a52cc1108a45d96ffd8dc0ab7f"
meta-oe           
meta-gnome        
meta-networking   
meta-multimedia   
meta-python       
meta-webserver    = "jethro:dc5634968b270dde250690609f0015f881db81f2"
meta-angstrom     = "angstrom-v2015.12-yocto2.0:6a31b67fbbe9e0ca9f597ab8b248d9fa980b156a"
meta-raspberrypi  = "jethro:f2cff839f52a6e6211337fc45c7c3eabf0fac113"
meta-fsl-arm      = "jethro:35b8b9bd9863de208ab60e33b55f10ee43e2619b"
meta-yocto-bsp    = "jethro:5b12268f6e17574999f91628a60e21711cf62ee4"
meta-qt5          = "jethro:d5536e34ec985c82b621448ab4325e5cbba38560"
```

I would be happy to provide further information as needed.

agherzan, 2016-02-20:
One interesting information would be what is your DISTRO_FEATURES.

zwerch, 2016-02-22:
Can you tell me where to find them?

zwerch, 2016-02-22:
We kind of fixed the issue with `udev` by changing it's position to a later one - `S20udev` worked for us.
But it still is very strange.

zwerch, 2016-02-25:
We didn't come across any other bugs so far, so this seems to be related to `udev` only. Maybe, to work around the issue, you could add a `udev_182.bbappend` containing this:
```bitbake
INITSCRIPT_PARAMS_udev = "start 20 S ."
```

You can close this now if you want.

agherzan, 2016-04-25:
Hm. I haven't seen this issue at all. I actually tested by creating a dummy init script in rcS which echoed something. It was started at next reboot and the message was shown.

closed 2016-04-25 by agherzan