agherzan/meta-raspberrypi #10: Device Tree Binary management for different kernels and machine configurations

toolmmy, 2016-03-15:
There are currently multiple issues regarding the device tree binary handling. Currently all dtb's are specified in the rpi-base.inc. But this causes multiple isses by different kernel recipes. For example the Kernel 3.18 doesn't provide the bcm2710-rpi-3-b.dtb

`
| make[3]: *** No rule to make target 'arch/arm/boot/dts/bcm2710-rpi-3-b.dtb'.  Stop.
| make[2]: *** [bcm2710-rpi-3-b.dtb] Error 2
| make[1]: *** [sub-make] Error 2
| make: *** [__sub-make] Error 2
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (log file is located at /builds/raspberrypi3/tmp/work/raspberrypi2-poky-linux-gnueabi/linux-raspberrypi/1_3.18.16+gitAUTOINC+1bb18c8f72-r0/temp/log.do_compile.24045)`

Same issue applies to the pitft22-overlay.dtb. This dtb is currently handled for each kernel version explicitly. New kernel version (such as linux-raspberrypi_4.4.bb) needs to add the patch-file again. 
`
| NOTE: make -j 8 overlays/pitft22-overlay.dtb
| make[3]: *** No rule to make target 'arch/arm/boot/dts/overlays/pitft22-overlay.dtb'.  Stop.
| make[2]: *** [overlays/pitft22-overlay.dtb] Error 2
| make[1]: *** [sub-make] Error 2
| make: *** [__sub-make] Error 2
| ERROR: oe_runmake failed
| ERROR: Function failed: do_compile (log file is located at /storage/builds/raspberrypi3/tmp/work/raspberrypi2-poky-linux-gnueabi/linux-raspberrypi/1_4.4.3+gitAUTOINC+a2c8f5b320-r0/temp/log.do_compile.27266)`

So in this case I would suggest a machine specific devite-tree handling

toolmmy, 2016-03-20:
@agherzan please checkout the changes made in here https://github.com/toolmmy/meta-raspberrypi/commit/cddd262b5904d464e6a751391edc7b41fc375907. If they are suitable for you I would create a pull request.

toolmmy, 2016-04-02:
Can be closed due to commit 75efa69

closed 2016-04-02 by toolmmy

agherzan, 2016-04-04:
Thanks @toolmmy