agherzan/meta-raspberrypi #11: bluetooth

danielnilsson9, 2016-03-27:
Hi

Any ideas for getting bluetooth to work on rpi3? I have tried patching bluez5 according to this issue:
https://github.com/raspberrypi/linux/issues/1314

and putting the image in /lib/firmware/brcm and all but I can't get it to work. I just get get timeouts when running hciattach, sometimes it says flashing firmware but then it times out again.

Anyone who got it working or is working on it that could help?

Thank you,
Daniel

toolmmy, 2016-04-09:
hi @danielnilsson9 
check out the following commit https://github.com/toolmmy/meta-raspberrypi/commit/e6e84cbb8dbee54c484105de8dc6c40169e881bb 
Make sure that you dont have a running agetty on ttyAMA0

danielnilsson9, 2016-04-09:
@toolmmy 

Looks pretty much identical to what I did, strange I couldn't get i working, definitely gonna try that out. Thank you for letting me know!

agherzan, 2016-04-13:
@toolmmy I see a set of patches on bluez5 but on trace of them online. Are these sent to bluez5? What is the status of these patches?

toolmmy, 2016-04-13:
@agherzan These patches regarding bluez5 seems to be very specific for the rpi3. Thats why they are provided by the foundation itself. I couldn't find any discussions in the bluez5 mailing list. Even when the patches are included in the upcoming versions of bluez5 we still rely on the latest versions provided by poky/openEmbedded which might can take time until they have spread out.

agherzan, 2016-04-25:
I merged a patch similar to what @toolmmy proposed. The only change is that the changes to bluez are rpi3 machine specific in the bbappend.

Cheers!

closed 2016-04-25 by agherzan