agherzan/meta-raspberrypi #14: Extends kernel configuration

giraldeau, 2016-04-15:
The bitbake build process configures the kernel using .cfg files. However, the meta-raspberrypi linux-rpi.inc creates a config file and calls make oldconfig. Is there a reason why the configuration is done this way? What would be the prefered way to extends the kernel configuration with additional parameters?

Thanks!

agherzan, 2016-04-25:
Hi. I was under the impression that we fixed this in the past by adding support for yocto config fragments. I haven't tested it in a while but I did it now and indeed it seems broken. We use the make oldconfig because we use a defconfig from the kernel tree repository.

yongli3, 2016-05-17:
@agherzan  I encountered this issue too. I cannot change the kernel config using the fragments cfg files(https://github.com/jumpnow/meta-rpi/issues/2). I found there is a patch about it: https://lists.yoctoproject.org/pipermail/yocto/2015-August/026101.html, can we merge it?

izak, 2016-07-25:
Ran into the same problem. Lost about a day trying to figure out why the patch works but the config is flatly ignored. Eventually applied the patch above and it appears to be the perfect solution.

toolmmy, 2016-07-25:
I'm not quite sure, bit it seems that the fragments are applied within the kernel_configme task (see kernel-yocto.inc addtask kernel_configme before do_configure after do_patch) before do_configure.

The do_configure of the linux-rpi.inc clears the complete .config in order to create a new one so all changes applied by the fragments are gone.

jenkins-ag, 2017-05-11:
test: please ignore

xkrys, 2017-06-19:
I'm also struggling with this, I want to add some support for can devices from my custom layers but changes from kernel fragments are ignored

agherzan, 2017-06-26:
There were some patches in the mailing list in this regards. If anyone has a little bit of time, a PR would be great.

ollyw, 2017-08-09:
Hi all, I tried the patches on the mailing list and extended for RPI3. I will put in a PR once it works, however, it doesn't seem to work fully, and I wondered if I could get some help?

I get a QA error that the /lib folder is not shipped. I can't see why the patches from https://lists.yoctoproject.org/pipermail/yocto/2015-August/026101.html would cause that issue as they don't seem to include changes to the FILES_ variables, and my fragment is compiling in kernel features rather than adding modules. Any pointers as to how to debug this would be great, then I can finish the PR. Apologies in advance, I am relatively new to yocto and new to the meta-raspberrypi layer custom kernel config stuff

betafive, 2017-10-25:
@agherzan, @giraldeau, @andig75 : Can this be closed now? Looks like the pull request above has been merged and should resolve this.

agherzan, 2017-10-27:
Yes. I personally tested this lately. Works just fine.

closed 2017-10-27 by agherzan

mardy, 2018-08-28:
This doesn't work for me. I'm trying to enable `CONFIG_SECURITY_APPARMOR` and a couple of other related options, but I don't see any of my changes in the final configuration (`tmp/work/raspberrypi3-pelux-linux-gnueabi/linux-raspberrypi/1_4.9.59+gitAUTOINC+e7976b2aff-r0/image/boot/config-4.9.59`). I do see my fragment (`apparmor.cfg`) being loaded in [tmp/work/raspberrypi3-pelux-linux-gnueabi/linux-raspberrypi/1_4.9.59+gitAUTOINC+e7976b2aff-r0/temp/run.do_kernel_metadata](https://paste.ubuntu.com/p/d74tTjpHsY/) and I see my changes in `tmp/work-shared/raspberrypi3/kernel-source/.meta/cfg/.config` but it looks like that it all in vain.

It looks like that that `.config` file is not being loaded, with `tmp/work/raspberrypi3-pelux-linux-gnueabi/linux-raspberrypi/1_4.9.59+gitAUTOINC+e7976b2aff-r0/linux-raspberrypi3-standard-build/.config` being loaded instead of it. Is this expected, am I doing something wrong?

kraj, 2018-08-28:
@mardy yes ```tmp/work/raspberrypi3-pelux-linux-gnueabi/linux-raspberrypi/1_4.9.59+gitAUTOINC+e7976b2aff-r0/linux-raspberrypi3-standard-build/.config``` is the final config which is applied during build.

mardy, 2018-08-29:
I found out the problem: my fragment contained `CONFIG_SECURITY_APPARMOR="y"`, but this was not included in the final config file because it depends on `CONFIG_SECURITY`, which is disabled in the default configuration. After adding also `CONFIG_SECURITY="y"`, it works.

Sorry for the noise! :-)