agherzan/meta-raspberrypi #16: Hardware Watchdog Pi 2

zwerch, 2016-04-22:
I read about a hardware watchdog kernel module [here](https://www.bayerschmidt.com/raspberry-pi/89-auto-reboot-a-hung-raspberry-pi-using-the-on-board-watchdog-timer.html).
I would like to have a hardware watchdog in my firmware on the Pi 2 B, too.
Is this supported at all? If yes, how can I enable it?

agherzan, 2016-04-22:
The module is compiled (tested on rpi3). You can load it with modprobe bcm2835_wdt .

zwerch, 2016-04-22:
Thanks for your answer!
Is there a way that I don't need to manually load the module but configure my image / whatever to load it automatically?

agherzan, 2016-04-22:
No worries. The way to do it in yocto is using KERNEL_MODULE_AUTOLOAD http://www.yoctoproject.org/docs/2.0/ref-manual/ref-manual.html#migration-1.7-kernel-module-autoloading .

zwerch, 2016-04-22:
Thanks a lot!

closed 2016-04-22 by zwerch

agherzan, 2016-04-22:
Welcome.