agherzan/meta-raspberrypi #17: Just got a Pi3 Model B in, image doesn't boot.

pauldotknopf, 2016-04-28:
I verified that NOOBS 1.9 boots correctly.

My Pi3 image (```MACHINE = "raspberrypi3"```) just shows 4 raspberries on boot.

I have also verified that my image for my Pi2 works (```MACHINE = "raspberrypi2"```).

Here is my setup.

```
MACHINE ??= "raspberrypi3"
DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_rpm"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
CONF_VERSION = "1"
LICENSE_FLAGS_WHITELIST = "commercial license"

```

```
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "raspberrypi3"
DISTRO            = "poky"
DISTRO_VERSION    = "2.0.1"
TUNE_FEATURES     = "arm armv7a vfp thumb neon callconvention-hard vfpv4 cortexa7"
TARGET_FPU        = "vfp-vfpv4-neon"
meta              
meta-yocto        
meta-yocto-bsp    = "HEAD:3b223f75eec1738fbc913858e8e11c8305e3edcb"
meta-raspberrypi  = "master:dedd75f08ffe3360722ab761da18987aa681edd1"
meta              = "jethro:4d879cb8d7384ac4a96f22c1664b8875f2d8f615"
meta-oe           
meta-python       
meta-networking   = "jethro:c305ac5d2f5285d5eec8952a4ca7f3b4f89aed96"
meta-camera-source = "master:62521e29499baf0be515e0b73ead80e34645fe33"
```

pauldotknopf, 2016-04-28:
If I copy over the ```bcm*``` files and ```bootcode.bin``` from NOOBS into the image created by yocto, it works.

agherzan, 2016-04-28:
Hi @pauldotknopf . What target are you baking?

pauldotknopf, 2016-04-28:
If I ```bitbake rpi-basic-image```, it happens.

agherzan, 2016-04-28:
After you burn the sdcard image, can you please tell what do you have in your first partition?

pauldotknopf, 2016-04-28:
```
-rw-r--r-- 1 pknopf pknopf   10849 Apr 28 05:58 bcm2708-rpi-b.dtb
-rw-r--r-- 1 pknopf pknopf   11128 Apr 28 05:58 bcm2708-rpi-b-plus.dtb
-rw-r--r-- 1 pknopf pknopf   12116 Apr 28 05:58 bcm2709-rpi-2-b.dtb
-rw-r--r-- 1 pknopf pknopf   12870 Apr 28 05:58 bcm2710-rpi-3-b.dtb
-rw-r--r-- 1 pknopf pknopf       0 Apr 28 05:58 bcm2835-bootfiles-20160421.stamp
-rw-r--r-- 1 pknopf pknopf   17924 Apr 28 05:58 bootcode.bin
-rw-r--r-- 1 pknopf pknopf      96 Apr 28 05:58 cmdline.txt
-rw-r--r-- 1 pknopf pknopf   36156 Apr 28 05:58 config.txt
-rw-r--r-- 1 pknopf pknopf    2512 Apr 28 05:58 fixup_cd.dat
-rw-r--r-- 1 pknopf pknopf    6484 Apr 28 05:58 fixup.dat
-rw-r--r-- 1 pknopf pknopf    9727 Apr 28 05:58 fixup_db.dat
-rw-r--r-- 1 pknopf pknopf    9729 Apr 28 05:58 fixup_x.dat
-rw-r--r-- 1 pknopf pknopf      60 Apr 28 05:58 image-version-info
-rw-r--r-- 1 pknopf pknopf 8978596 Apr 28 05:58 kernel7.img
drwx------ 2 pknopf pknopf    2048 Apr 28 05:58 overlays
-rw-r--r-- 1 pknopf pknopf  616248 Apr 28 05:58 start_cd.elf
-rw-r--r-- 1 pknopf pknopf 4892168 Apr 28 05:58 start_db.elf
-rw-r--r-- 1 pknopf pknopf 2743544 Apr 28 05:58 start.elf
-rw-r--r-- 1 pknopf pknopf 3844392 Apr 28 05:58 start_x.elf
```

pauldotknopf, 2016-04-28:
In case you want to inspect (or boot) the image, here it is.

[rpi-basic-image-raspberrypi3.rpi-sdimg](https://drive.google.com/file/d/0By5zzSLiMe2INjlySUQ3UEtJSVk/view?usp=sharing)

pauldotknopf, 2016-05-02:
Any ideas with this?

toolmmy, 2016-05-02:
I see a lot of different layers. did you try just poky with meta-raspberrypi? Ive tested the latest release (krogoth) yesterday without any issues (just the rpi-sd-card but anyway)

agherzan, 2016-05-04:
I tested rpi3 and rpi2 and they both seem find. Could you check what @toolmmy is suggesting?

pauldotknopf, 2016-05-04:
I have, and it still didn't work. All my branches were on master.

I will do it again, and share my layer info again, removing the ones that aren't needed for the basic image.

pauldotknopf, 2016-05-04:
Have you tried to boot the image I shared with you? I wonder if it is just my batch of rpi s. This happens on all 5 of mine.

agherzan, 2016-05-04:
Let me try it now. Be right back.

agherzan, 2016-05-04:
Indeed. It doesn't boot for me too (using the image you provided).

pauldotknopf, 2016-05-04:
Ya, well, I'll try the latest release. I still don't think it will work though. I will let you know.

tmpsantos, 2016-05-05:
Also got the four berries with my Pi3:

```
meta-yocto-bsp    = "jethro:6dba9abd43f7584178de52b623c603a5d4fcec5c"
meta-raspberrypi  = "master:9912d38e97671704822d1aa05312a0439cb650d3"
```

Gonna try with `krogoth` now.

tmpsantos, 2016-05-05:
Got stuck on the 4 berries with `krogoth` too.

```
meta-yocto-bsp    = "krogoth:898a78357e72fb80f1f8e26ba90bad5b7b054a4f"
meta-raspberrypi  = "master:9912d38e97671704822d1aa05312a0439cb650d3"
```

brunoabinader, 2016-05-05:
> If I copy over the bcm* files and bootcode.bin from NOOBS into the image created by yocto, it works.
 👍 1  

I had success by replacing `bcm2710-rpi-3-b.dtb` specifically.

pauldotknopf, 2016-05-05:
@brunoabinader, ya, I can't say I try each individual file specifically to see what worked. Just copied a bunch of files that looked relevant, and it worked! ;)

tmpsantos, 2016-05-06:
The firmware binary on NOOBS 1.9 can be found at `c230b2b9b5a38615a4ab17539c3b438064fe5d83` of https://github.com/raspberrypi/firmware.git The `firmware` recipe on `meta-raspberry` master is a few commits ahead.

tmpsantos, 2016-05-06:
Just realized that `bcm2710-rpi-3-b.dtb` is built from the kernel sources and has nothing to do with the firmware binaries.

I just tried building with kernel 4.4 and also didn't help.

agherzan, 2016-05-13:
I'm wondering why can't I reproduce this at all... As well the meta-rpi bsp layer is used in resin product and works fine for all the boards rpi1, rpi2 and rpi3. So the only problematic file is the dtb?

pficheux, 2016-06-05:
Replaced bcm2710-rpi-3.dtb, but does not work :(

tmpsantos, 2016-06-05:
> Replaced bcm2710-rpi-3.dtb, but does not work :(

Where are you taking the firmware from?

agherzan, 2016-06-06:
Are you sure the board doesn't boot? It might be that there is no serial console configured and just don't see any feedback. How do you test if the board actually boots?

pficheux, 2016-06-06:
I've copied it from Raspbian (a rough test...). The same configuration (PC + Adafruit USB cable + Pi3) works with Raspian by adding the following line to config.txt :

dtoverlay=pi3-disable-bt

(from https://openenergymonitor.org/emon/node/1231.

agherzan, 2016-06-06:
Can you try to configure the serial as instructed in the README file? https://github.com/agherzan/meta-raspberrypi/blob/master/README#L242

pficheux, 2016-06-06:
OK, sorry for not reading the README...

agherzan, 2016-06-06:
It's fine. It's a long file and one can easily miss information from there. I'm thinking of having a normal documentation soon.

pficheux, 2016-06-06:
Works fine with ENABLE_UART = "1 !

Regards

agherzan, 2016-06-06:
Super cool! Glad we sorted out.

zorce, 2016-06-07:
Hi,
Did you solve the issues? I have the same kind of issues. Cannot boot the rpi-basic-image just end up with 4 berrys. If I copy the dtb files from an existing known to work image I end up with the rainbow screen instead...

agherzan, 2016-06-07:
> Can you try to configure the serial as instructed in the README file? https://github.com/agherzan/meta-raspberrypi/blob/master/README#L242

pficheux, 2016-06-07:
I fixed the problem with ENABLE_UART = "1", but I use a serial console, no HDMI output.

zorce, 2016-06-08:
I will try that need to get an FTDI usb to serial cable.
But shouldn't I get HDMI output?

pficheux, 2016-06-08:
I use this USB cable https://www.adafruit.com/product/954. i didn't test HDMI but cmdline.txt contains "console=serial0,115200"  ->  console should be on serial port only (?)

agherzan, 2016-06-08:
That is not configured by default. You can add as many console= as you wish. For HDMI you need to add console=tty1.

zorce, 2016-06-08:
Thank, will check this when I get home!

agherzan, 2016-06-14:
Will close this for now. If you still encounter problems with RPI3 feel free to comment.

closed 2016-06-14 by agherzan

Anhunghezo, 2022-03-15:
I got the same error, I tried all way above and not success.

agherzan, 2022-03-15:
@Anhunghezo We'd need a bit more information about your setup. How do you check if the board boots? What do you expect? Did you configure the serial and check it? How can someone reproduce this behaviour?