agherzan/meta-raspberrypi #18: Feature request: configuration parameter for freeing up /dev/ttyAMA0 on RPi2

nowls, 2016-05-10:
I'm running an Adafruit GPS Hat on a Raspberry Pi 2. The pins on that product are mapped to the TXD0/RXD0 pins and should be available through /dev/ttyAMA0. Is there a way I can use a dtoverlay option to override "console=serial0,115200", skip the firmware's search-and-replace, and just shove "console=tty1" in there to free up /dev/ttyAMA0? I believe this should be a configuration parameter I could change in a bbappend in my own OpenEmbedded layer. This is related to the discussion here: https://github.com/RPi-Distro/repo/issues/22

agherzan, 2016-05-10:
That is actually a very good point. I know resin is doing something similar to address this issue. Here is the relevant commit: https://github.com/resin-os/resin-raspberrypi/commit/12f0123b5d536102da33284aff33b37ae2bbbd29

So you can always overwrite CMDLINE with your own configuration.

agherzan, 2016-06-14:
@nowls did you manage to get it fixed?

nowls, 2016-06-15:
Sort of... I didn't do it in a general way. I ended up making my recipes-kernel/linux/linux-raspberrypi_%.bbappend that uses Sed to replace serial0,115200 with tty1. I'm still learning openembedded magic...

closed 2017-01-20 by agherzan