agherzan/meta-raspberrypi #19: Taskhash mismatch

pficheux, 2016-06-09:
Hi,

I've got strange problem about 'taskhash mismatch'  when I try to rebuild core-image-minimal after a 'cleansstate'.

ERROR: core-image-minimal-1.0-r0 do_image_rpi_sdimg: Taskhash mismatch 1b48d92377c7cedb58231830828f590b verses 92b3b86f0e026e3159ba0f71477c092a for /media/pierre/SAMSUNG_EXT4/Yocto/krogoth/meta/recipes-core/images/core-image-minimal.bb.do_image_rpi_sdimg
ERROR: Taskhash mismatch 1b48d92377c7cedb58231830828f590b verses 92b3b86f0e026e3159ba0f71477c092a for /media/pierre/SAMSUNG_EXT4/Yocto/krogoth/meta/recipes-core/images/core-image-minimal.bb.do_image_rpi_sdimg

Looks like the  commit https://github.com/agherzan/meta-raspberrypi/commit/cd338c26e25640cd4d67d17646684e3d3314dc74 added the following line to classes/sdcard_image-rpi.bbclass :

IMAGE_CMD_rpi-sdimg[vardepsexclude] = "IMAGEDATESTAMP"

If I comment the line, everything works fine... Actually i should admit I don't really understand the use of that line (and the Yocto documentation):

vardepsexclude: Specifies a space-separated list of variables that should be excluded from a variable's dependencies for the purposes of calculating its signature.

regards

agherzan, 2016-06-09:
There is a patch for this and I will take a look on it this week. In the meanwhile you can check the mailing list.

agherzan, 2016-06-14:
This issue was fixed in master. Please check and reopen this issue if you find it otherwise.

agherzan, 2016-06-14:
Can you recheck with current poky master? I can't replicate the issue anymore.

pficheux, 2016-06-15:
OK I will do that today.

pficheux, 2016-06-20:
Hi,
Look good now, no more error.

agherzan, 2016-06-20:
Cool. Closing.

closed 2016-06-20 by agherzan