agherzan/meta-raspberrypi #20: Lirc-Rpi and w1-therm modules not populating /dev /sys

wilberforce, 2016-06-14:
I'm struggling with one wire temperature sensor and an Ir receiver using a recent yocto build, here is what I'm seeing on a Jessie build:

[    4.704991] i2c /dev entries driver
[    5.484709] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    5.821942] systemd-udevd[96]: starting version 215
[    6.350574] lirc_dev: IR Remote Control driver registered, major 245
[    6.399151] lirc_rpi: module is from the staging directory, the quality is unknown, you have been warned.
[    6.407353] Driver for 1-wire Dallas network protocol.
[    6.420098] w1-gpio onewire@0: gpio pin 4, external pullup pin -1, parasitic power 0
[    6.420157] w1_add_master_device: set_pullup requires write_byte or touch_bit, disabling
[    6.521092] gpiomem-bcm2835 20200000.gpiomem: Initialised: Registers at 0x20200000
[    7.373714] bcm2835-rng 20104000.rng: hwrng registered
[    7.840786] lirc_rpi: auto-detected active low receiver on GPIO pin 24
[    7.841199] lirc_rpi lirc_rpi: lirc_dev: driver lirc_rpi registered at minor = 0
[    7.841222] lirc_rpi: driver registered!

not working yocto:
[    6.412996] i2c /dev entries driver
[    6.479388] Driver for 1-wire Dallas network protocol.
[    6.564720] lirc_dev: IR Remote Control driver registered, major 244
[    6.617053] lirc_rpi: module is from the staging directory, the quality is unknown, you have been warned.
[    6.624019] lirc_rpi: gpio chip not found!

I have device tree entries in /boot/config.txt

dtoverlay=w1-gpio
dtparam=i2c_arm=on
dtoverlay=i2c-rtc,ds1307
dtoverlay=lirc-rpi,gpio_out_pin=17,gpio_in_pin=24

I can load the modules by using modprobe, or adding to /etc/modules, however the modules don't seem to populate /sys/bus/w1 with the actual devices, and the lirc-rpi fails to load with gpio chip not found.

Any clues? Thanks

agherzan, 2016-06-14:
Sadly I hit a similar issue lately too (https://github.com/resin-os/resin-raspberrypi/issues/24). I'm going to investigate it very soon as this hits other overlays too as I see. You might want to try the latest kernel 4.4 and some patches for dtbo support too that are already in the mailing list. I'll get to them soon and see what's going on.

wilberforce, 2016-06-15:
Thanks for the reply. 

I'm using 4.4.9:
```
root@raspberrypi:~# uname -a
Linux raspberrypi 4.4.9 #1 Mon Jun 6 18:32:35 NZST 2016 armv6l armv6l armv6l GNU/Linux
```
I've tried looking at udev rules, but that didn't work for me. I'm not sure what debugging I can added - there is the device tree debug flag, however the tools aren't part of the yocto build

wilberforce, 2016-06-15:
It works for some overlays and not others.

```
root@raspberrypi:~# cat /media/card/config.txt
disable_overscan=1
gpu_mem_256=16
gpu_mem_512=16
gpu_mem_1024=16
dtoverlay=w1-gpio,gpiopin=4
dtparam=i2c_arm=on
dtoverlay=i2c-rtc,pcf8563
dtparam=act_led_trigger=heartbeat
dtoverlay=lirc-rpi
arm_freq=900
```

working:
```
dtparam=i2c_arm=on
dtparam=act_led_trigger=heartbeat
```
The modules are loaded and the led blinks a heartbeat (useful to tell when to pull the power when headless)

`i2c-dev` is still required in `/etc/modules` to get the i2c devices (same with Jessie)

not working:
```
dtoverlay=w1-gpio,gpiopin=4
dtoverlay=i2c-rtc,pcf8563
dtoverlay=lirc-rpi
```

interestingly, I've never managed to the rtc overlays working on jessie either.

> some patches for dtbo support too that are already in the mailing list.
Sorry, not clear on which list to look at, as there are quite a few! 

I found this:
http://patchwork.openembedded.org/patch/123747/

It think is is related to the filename extensions - not sure it applies to this case?

wilberforce, 2016-06-16:
Using the debug=1 param shows the issue for the lirc case

modprobe lirc_rpi debug=1

```
lirc_dev: IR Remote Control driver registered, major 252 
lirc_rpi: module is from the staging directory, the quality is unknown, you have been warned.
lirc_rpi lirc_rpi.0: lirc_dev: driver lirc_rpi registered at minor = 0
lirc_rpi: driver registered!
lirc_rpi: is_right_chip bcm2708_gpio 0
lirc_rpi: to_irq 103
lirc_rpi: auto-detected active low receiver on GPIO pin 18
```

And the yocto case

```
modprobe lirc_rpi debug=1
Is_right_chip pin_ctrl-bcm2835 -1
```

So it's the pin_ctrl version of the driver not matching!

wilberforce, 2016-06-16:
This is probably out of date...

https://github.com/raspberrypi/linux/issues/936#issuecomment-109802801

wilberforce, 2016-07-07:
@agherzan 
> I'm going to investigate it very soon as this hits other overlays too

Have you had a chance to look yet?

I thought it might be related to this 

https://github.com/agherzan/meta-raspberrypi/commit/8a21c1dda6150ae2596f3932ffbc6cb317ebcdf8

However this didn't appear to fix :-(

agherzan, 2016-07-08:
@wilberforce Indeed we found the issue in my case. It was because of a typo in overlayers directory. https://github.com/resin-os/resin-raspberrypi/commit/ca41c2efc49983fe3452aecde2bdb4d57b33cc2c which seems to be unrelated to your problem. I personally don't have an idea what might cause this right now. 

You ar eusing a jessy build as I see. Can you use current master?

wilberforce, 2016-07-26:
looks like the sources were out of sync.

` bitbake -c cleansstate virtual/kernel`

This cleared out, and a rebuild shows:

```
root@raspberrypi2:~# lsmod
Module                  Size  Used by
ipv6                  350058  24
mt7601u                74895  0
mac80211              538840  1 mt7601u
cfg80211              436606  2 mac80211,mt7601u
evdev                  11396  0
rfkill                 16902  1 cfg80211
w1_therm                3520  0
i2c_bcm2708             4770  0
lirc_rpi                6506  0
bcm2835_gpiomem         3036  0
bcm2835_wdt             3225  0
w1_gpio                 3657  0
wire                   25280  2 w1_gpio,w1_therm
cn                      4374  1 wire
lirc_dev                8438  1 lirc_rpi
uio_pdrv_genirq         3164  0
rc_core                16404  1 lirc_dev
uio                     8256  1 uio_pdrv_genirq
```

```
root@raspberrypi2:~# cat /sys/devices/w1_bus_master1/28-000006b3ab40/w1_slave
38 01 4b 46 7f ff 08 10 18 : crc=18 YES
38 01 4b 46 7f ff 08 10 18 t=19500
```

closed 2016-07-26 by wilberforce