agherzan/meta-raspberrypi #21: not building anymore with latest krogoth yocto branch

LocutusOfBorg, 2016-07-01:
```
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Error, the PACKAGE_ARCHS variable (all any noarch armv5hf-vfp armv5thf-vfp armv5ehf-vfp armv5tehf-vfp armv6hf-vfp armv6thf-vfp armv7ahf-vfp armv7at2hf-vfp armv7vehf-vfp armv7vet2hf-vfp armv7vehf-neon armv7vet2hf-neon armv7vehf-neon-vfpv4 armv7vet2hf-neon-vfpv4 cortexa7hf-vfp cortexa7hf-neon cortexa7hf-neon-vfpv4 cortexa7t2hf-vfp cortexa7t2hf-neon cortexa7t2hf-neon-vfpv4 raspberrypi2) for DEFAULTTUNE (cortexa7thf-neon-vfpv4) does not contain TUNE_PKGARCH (cortexa7hf-neonvfpv4).
```

the culprit is change: 31351ce146bf9c61d779456cd68ff7c2c9d4c8a8

LocutusOfBorg, 2016-07-01:
reverting http://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=31351ce146bf9c61d779456cd68ff7c2c9d4c8a8 this commit make it work again, but upstream told it might be an issue of your layer

agherzan, 2016-07-08:
Hi @LocutusOfBorg this seems to be an issue that got solved in the meanwhile as trying now with current master seems to not reproduce your issue. I used:

```
Build Configuration:
BB_VERSION        = "1.31.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "raspberrypi"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1+snapshot-20160708"
TUNE_FEATURES     = "arm armv6  vfp arm1176jzfs callconvention-hard"
TARGET_FPU        = "hard"
meta              
meta-poky         
meta-yocto-bsp    = "master:4e889c04ba518dac492c193224a790a877f43f40"
meta-raspberrypi  = "master:43ecd9ac992216974316f954683547e5867c5612"
meta-oe           
meta-python       
meta-networking   
meta-multimedia   = "fixes:9379f00022a1a538d4bc63b90a1ae9642d949953"
meta-qt5          = "master:c16c4662db7947ce48af35e0b93a440fe955588a"
```

LocutusOfBorg, 2016-07-08:
yes, fixed here
http://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=e6917603e2a7ad8a920d3883d2ddf196b6038dc5

G.

closed 2016-07-08 by LocutusOfBorg

agherzan, 2016-07-08:
Cool. Thanks man.