agherzan/meta-raspberrypi #26: enable bluetooth

my007ms, 2016-08-09:
Hello,

I am trying to enable bluetooth in my Pi3 image so now i have 

local.conf

MACHINE ??= "raspberrypi3"
IMAGE_FSTYPES = "rpi-sdimg"
IMAGE_INSTALL_append = "bluez5 wireless-tools linux-firmware-brcm43430 wpa-supplicant rpm"
DISTRO_FEATURES_append += " bluez5 bluetooth wifi systemd"
ENABLE_UART = "1"
MACHINE_FEATURES += " pi3-miniuart-bt"

bblayers.conf 

BBLAYERS ?= " \
  /home/yocto/poky/meta \
  /home/yocto/poky/meta-poky \
  /home/yocto/poky/meta-yocto-bsp \
  /home/yocto/poky/meta-raspberrypi \
  /home/yocto/poky/meta-openembedded/meta-oe \
  "

hcitools does not show any bluetooth interface 

and if i try set 

KERNEL_DEVICETREE = "overlays/pi3-miniuart-bt-overlay.dtb"
in
meta-raspberrypi/recipes-core/images/rpi-basic-image.bb
or 
local.conf 

in both case i get rainbow screen in my HDMI screen and freeze after that 

and if i try 
Set KERNEL_DEVICETREE (in conf/machine/raspberrypi.conf) it's ignored the value whatever it is

i try following 
https://github.com/raspberrypi/firmware/issues/553
and 
https://github.com/RPi-Distro/repo/issues/22 

and no luck 

i don't want or use any of GPIO pin or serial so i don't mind disable it all if this will fix the bluetooth issue 

Thanks

toolmmy, 2016-08-10:
Hi,

the pi3-miniuart-bt is reducing the Baudrate to the serial hci interface to 115200 baud in order to manage the serial consoke in parallel. The curren setup is using the actual baudrate for the hci interface 9xxxxx. Could you check your setup without the rpi3-minicom-bt overlay?

my007ms, 2016-08-10:
Hi,

without rpi3-minicom-bt overlay it boot fine with HDMI console but no bluetooth in hciconfig -a
but if i add extra USB bluetooth it work fine only the built in bluetooth never show up

Thanks

toolmmy, 2016-08-18:
Hi, 

are you using a systemd environment? Otherwise you need a sysvinit script that performes the hciattach during system start. Could you please try the following command manually

/usr/bin/hciattach -n /dev/ttyAMA0 bcm43xx 921600 noflow -

my007ms, 2016-08-18:
Thanks toolmmy 

Yes it's work i was think that udev should take care of create the dev
I try with systemd and it work fine also i try the manual command and it work fine as well 

However in both cases SCO socket not working ( it's received zero packet ) this not only in yacto image i think in all Pi Images including NOOBS and RASPBIAN, have you by any chance try used SCO sockets in your side with buit-in bluetooth ?

Thanks

closed 2016-08-23 by agherzan

sherifomran, 2018-04-09:
did you change audio = on in /boot/config.txt ?
this will enable the alsa device otherwise you wont detect it.