agherzan/meta-raspberrypi #30: do_image_rpi_sdimg failed  oe-core in krogoth branch

mickey-happygolucky, 2016-09-27:
```
| DEBUG: Executing python function set_image_size
| DEBUG: Python function set_image_size finished
| DEBUG: Executing shell function do_image_rpi_sdimg
| Creating filesystem with Boot partition 40960 KiB and RootFS 90112 KiB
| dd: failed to open '/rpi-basic-image-raspberrypi3-20160927134756.rootfs.rpi-sdimg': Permission denied
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_image_rpi_sdimg (log file is located at /home/mickey/work/yocto/rpi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/rpi-basic-image/1.0-r0/temp/log.do_image_rpi_sdimg.7703)
ERROR: Task 16 (/home/mickey/work/yocto/rpi/build/../poky/meta-raspberrypi/recipes-core/images/rpi-basic-image.bb, do_image_rpi_sdimg) failed with exit code '1'
```

do_image_rpi_sdimg is failed after 4817e2c087097c02755d6309304878e42cf61d3c commit.
I reverted 4817e2c087097c02755d6309304878e42cf61d3c for test, then do_image_rpi_sdimg is succeeded.

This commit is correspond on oe-core in master branch. but it is unnecessary for the krogoth branch.

By the way, why meta-raspberrypi does not have krogoth branch?

sherpya, 2016-10-11:
even reverting the commit I get `| /home/sherpya/poky/build/rpi3/tmp/deploy/images/raspberrypi3/Image-hifiberry-amp.dtbo: No such file or directory` (no dtbo support in krogoth)

rguenthe, 2016-10-12:
Same problem here using the master branch. 
The devicetree overlay files have extension `dtbo.dtb` instead of `dtbo`. As a workaround I changed `DTB_EXT=${DTB##*.}` to `DTB_EXT="dtb"` (Line 119) in `classes/sdcard_image-rpi.bbclass` but this should rather be fixed by using correct file names (kernel issue?)

lurch, 2016-10-12:
IIRC whether you get `.dtb` or `.dtbo` files, depends on which version of the kernel you're using?

mickey-happygolucky, 2016-10-12:
I know that fixed some problems in master branch. 
however I want to use krogoth branch without some workarounds. 

So, I tought that I wish meta-raspberrypi had krogoth branch like a jethro, fido, dylan...

If maintenance cost is too much, I give up that use the oe-core in krogoth branch.

agherzan, 2016-10-12:
Indeed we included changes that are not compatible anymore with krogoth. I will try to create a krogoth branch.

agherzan, 2016-10-12:
I create a krogoth branch which should work fine with yocto/meta-oe krogoth. Will close this for now. Please tell me if you still encounter issues.

closed 2016-10-12 by agherzan

dl9pf, 2017-01-11:
I see the same issue on morty branch, too.

agherzan, 2017-01-18:
@dl9pf What revision of meta-rpi and poky are you using?

dl9pf, 2017-01-18:
Problem was in custom image class that did not catch-up with the changes in morty. Found/Solved. Sri for the noise.

agherzan, 2017-01-18:
Thanks @dl9pf

Larswad, 2017-03-29:
Bumping here! What was the solution to this? I don't get it. I get the same problem with morty and sdcard image generation for raspberry. The overlay naming is off in two ways, should I patch the image class or checkout another version of it?

agherzan, 2017-04-03:
@Larswad what revision of meta-raspberrypi and poky are you using?

Larswad, 2017-04-03:
Thanks, agherzan for responding! Well, I followed this guys guide http://www.jumpnowtek.com/rpi/Raspberry-Pi-Systems-with-Yocto.html
So it's morty,
It seems the revision for meta-raspberrypi is is 4.9.10 and the commit id cce6292e41493158c26f5b1b7fded97faacf10d7

For poky its:
DISTRO_VERSION = "2.2.1"
DISTRO_CODENAME = "morty
Commit id: 924e576b8930fd2268d85f0b151e5f68a3c2afce

I did checkout of 4817e2c and then it built the sdcard image, but it seems strange to me that head commit didn't work.

agherzan, 2017-04-05:
I'm confused. Sorry. So what combination of poky and meta-raspberrypi failed for you?

Larswad, 2017-04-07:
The platform and meta-raspberry pi i described above failed this way.

EXCEPT, for when I then checked out 487e2c of meta-raspberrypi of the bbclass file for building sdcard image.

Now I've reverted the whole thing and I'm going for a krogoth build instead (so I'm happy).
But yet still, I can't understand why the head commit of meta-raspberrypi in morty didn't work regarding this exact issue with the dtb0 vs dtb files.

agherzan, 2017-04-17:
I just checked morty meta-rpi + morty poky and it worked just fine.