agherzan/meta-raspberrypi #31: RPi 3 - wifi ready image

mark2b, 2016-10-03:
What is a way to get wifi ready image for RPi3?
There is no wlan0 interface for build-in wifi

rguenthe, 2016-10-06:
Somehow the firmware for the wifi module is not getting installed on the rootfs, even though the firmware package is appended to the `MACHINE_EXTRA_RRECOMMENDS` in `raspberrypi3.conf`.
Adding `IMAGE_INSTALL += "linux-firmware-brcm43430"` to my image recipe worked for me.

I checked the variable with `bitbake -e` and it seems to be set correctly ...
`...MACHINE_EXTRA_RRECOMMENDS=" linux-firmware-brcm43430  kernel-modules"...`

rguenthe, 2016-10-06:
found the reason why installing the firmware using the `MACHINE_EXTRA_RRECOMMENDS` variable is not always working:
> This variable affects only images based on packagegroup-base, which does not include the core-image-minimal or core-image-full-cmdline images.

(Source: Yocto Documentation: http://www.yoctoproject.org/docs/current/ref-manual/ref-manual.html#var-MACHINE_EXTRA_RRECOMMENDS)

agherzan, 2016-10-14:
The problem with this is that `MACHINE_EXTRA_RRECOMMENDS` is in packagegroup-machine-base which is brought in by core-image. While this being true, the included images templates in meta-raspberrypi are based on core-image-minimal which does not include in turn `MACHINE_EXTRA_RRECOMMENDS`. I'm thinking of bringing in `linux-firmware-brcm43430` in rpi-basic-image. I think that would make sense. But generally, user should create their own images or extend the ones included.

toolmmy, 2016-10-26:
I guess it is important to know that the broadcom-firmware is now part of the linux-firmware of morty (path: root/meta/recipes-kernel/linux-firmware/linux-firmware_git.bb). The funny thing is that it is called bcm43430 instead of brcm43430 :)

agherzan, 2016-10-31:
That is indeed interesting @toolmmy . I would expect a sysroot conflict but I haven't seen it on master (which should have this change).

Minipada, 2016-12-10:
I compiled a new image, with master repos and I couldn't detect wifi until I included `brcmfmac43430-sdio.txt`.
It seems it was provided before (at least in jethro) by the current layer but since it is now handled by the standard poky repo, it is not handled here. They'd need to patch this to have a wifi ready image

agherzan, 2017-01-21:
@Minipada Morty should have the firmware included in the test image (rpi-test-image). It includes connman too so you can check connectivity. I'll close this issue for now. Feel free to open it if needed.

xpgdk, 2017-02-23:
@agherzan I believe linux-firmware in Poky contains the actual bcm43430 firmware (brcmfmac43430-sdio.bin). However, the NVRAM configuration file (brcmfmac43430-sdio.txt) does not seem to be present. I think that's the way it's supposed to be, as the NVRAM file seems to be highly board specific.
So, the NVRAM file should probably be part of meta-raspberrypi.

agherzan, 2017-03-01:
@xpgdk Correct. Could you push a patch which would bbappend the nvram configuration too?

agherzan, 2017-03-01:
As well it would be great to send this patch to master and morty.

gizero, 2017-03-08:
Hi! Anyone working on this? I'd like to help fixing it, but I still don't have a clear view of where things should be fixed (pretty new to RPi boot/configuration logic). I'm building on master and my image `inherit core-image`. 
I see no wlan0 ifce and, from `dmesg` I get:
```
brcmfmac_sdio mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.txt failed with error -2
```

xpgdk, 2017-03-08:
@gizero I have a local solution, which I am working on getting pushed. 
I'm just lacking the time to do it, but haven't forgot it...

gizero, 2017-03-16:
@xpgdk a patch for this seems to be already staged in @kraj repo. I'm baking the thing to give it a test... just thought to leave a comment to avoid duplicating the efforts

agherzan, 2017-03-21:
Current master has this fixed. Ping back if that is not working for you.

closed 2017-03-21 by agherzan

iunu-build, 2017-03-21:
Not seeing any commit in master with this file or a bbappend, just omxplayer update.

gizero, 2017-03-22:
Commit 6ce6e57782c23fd21ad47286ac8b809ce989ea78 is doing the job!

bog-dan-ro, 2017-04-01:
I just test it using master branch and this issue is not fixed. brcmfmac43430-sdio.txt is not present in my core-image-base.

I also added the following vars to my local.conf
IMAGE_INSTALL_append += " linux-firmware iw networkmanager"
DISTRO_FEATURES_append += " wifi"

agherzan, 2017-04-03:
@bog-dan-ro The firmware you need is installed in `${PN}-bcm4343` . Ref.: https://github.com/agherzan/meta-raspberrypi/blob/master/conf/machine/raspberrypi3.conf#L8

bog-dan-ro, 2017-04-08:
@agherzanI've seen that line, but, sadly, is not installed :)

agherzan, 2017-04-17:
@bog-dan-ro what exact revision are you using and what machine?

agherzan, 2017-04-17:
@bog-dan-ro I pushed a new set of patches for morty. Those should fix your issue and bring back WiFi in morty.

fwag, 2017-08-07:
Anyone has tried on new pyro ? I'm facing this issue
[ 3.392744] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2

agherzan, 2017-08-23:
What revision are you using?

marvic41, 2017-12-17:
Hello,
Got error while compiling for Rpi zero w using Main branch, 
Added in local.conf
`IMAGE_INSTALL_append += linux-firmware-brcm43430`
Error
`ERROR: Nothing RPROVIDES 'linux-firmware-brcm43430' (but /home/kkk/poky/meta-raspberrypi/recipes-core/images/rpi-hwup-image.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'linux-firmware-brcm43430' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['linux-firmware-brcm43430']
ERROR: Required build target 'rpi-hwup-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['rpi-hwup-image', 'linux-firmware-brcm43430']`

How to fix this.

fwag, 2017-12-17:
Are you also using meta-rpi ?

Inviato da iPhone

marvic41, 2017-12-17:
Yes. tried with both branches
`git clone https://github.com/mirzak/meta-raspberrypi/ -b rpi-zero-wifi-support
~/poky/build$ git clone git://git.yoctoproject.org/meta-raspberrypi`

xuhcc, 2017-12-17:
@marvic41 

Recipe name should be different: **linux-firmware-bcm43430** (without **r**). Also, `+=` is not needed when you use `_append`.

Try this: `IMAGE_INSTALL_append = "  linux-firmware-bcm43430 "`

marvic41, 2017-12-18:
@xuhcc
Thanks, Yes without r it compiled.