agherzan/meta-raspberrypi #35: bitbake core-image-sato failed with oe-core master or morty

mickey-happygolucky, 2016-11-15:
I tried to bitbaking core-image-sato.
However bitbake showed error message as follows.

```txt
bitbake core-image-sato
WARNING: Host distribution "Ubuntu-16.10" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |############################################| Time: 0:00:00
Loaded 1338 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'linux-firmware-brcm43430' (but /home/mickey/work/yocto/rpi-morty/poky/meta/recipes-core/packagegroups/packagegroup-base.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'linux-firmware-brcm43430' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['linux-firmware-brcm43430']
NOTE: Runtime target 'packagegroup-base-extended' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-base-extended', 'linux-firmware-brcm43430']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-base-extended', 'linux-firmware-brcm43430']

Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

I suspected commit 380740d866b5360647fb5980cb8df69f8af814e5.
I tried bitbake after reverting this commit.
Then bitbake is succeeded.

Have you seen problems like this?

agherzan, 2016-11-16:
Hi Yusuke,

We fixed this in master. Feel free to give it a try.

mickey-happygolucky, 2016-11-16:
Hi Andrei,

I confirmed. It works so fine.
Thanks!

closed 2016-11-16 by mickey-happygolucky