agherzan/meta-raspberrypi #36: Problem with firmware files

mirzak, 2016-11-16:
Hi guys.

Anyone else experiencing problems booting on current HEAD?

I am currently testing on a raspberrypi2 and it builds just fine but does not boot. Tested both Linux and u-boot. I tried using the "bootfiles" from krogoth branch (just copied them to boot section on SD) and that worked! Though I did not use the device tree from krogoth but from HEAD, the krogoth one did not work for me. So there seem to some conflicts here, or is it just me?

I have also tested building krogoth branch and that works just fine.

I noticed that you in a recent commit changed to use tarballs for firmware [1] and that you changed the firmware date there, was this intentional? I shall admit that I do not quite understand the dependencies of the firmware files, would love if you could explain that a bit or point me to some documentation.

[1]. http://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/commit/?id=716b6a9cd7f24a8bacd539bb40519d185e3f963a

mirzak, 2016-11-27:
I took the matter in my own hands and did some digging. It is all related to https://github.com/raspberrypi/firmware/commit/d0bc6ce8e2ae7850959fed4edb0695f3cddfb96a. Which changed the default clock rate of UART0 (console on most pi`s).

So the device actually boots, but the console serial port does not work properly.

agherzan, 2016-12-06:
Hi @mirzak . So this means you are using uboot, right?

mirzak, 2016-12-06:
Hi @agherzan. Yes that is true, I am using u-boot.

agherzan, 2017-01-20:
@mirzak Current morty branch should have this fixed. Feel free to reopen if needed.

closed 2017-01-20 by agherzan