agherzan/meta-raspberrypi #43: u-boot build failed in morty branch

mickey-happygolucky, 2017-01-24:
I checked out  morty branch both poky and meta-raspberrypi.
Then I added [KERNEL_IMAGETYPE = "uImage"] to my local.conf and ran bitbake.
but I got an error such as follows.

```txt
| board/raspberrypi/rpi_2/built-in.o: In function `board_get_usable_ram_top':
| /home/mickey/work/yocto/rpi-morty/build_u-boot/tmp/work/raspberrypi3-poky-linux-gnueabi/u-boot/v2016.03+gitAUTOINC+df61a74e68-r0/git/board/raspberrypi/rpi_2/../rpi/rpi.c:272: undefined reference to `fw_dtb_pointer'
| board/raspberrypi/rpi_2/built-in.o: In function `set_serial_number':
| /home/mickey/work/yocto/rpi-morty/build_u-boot/tmp/work/raspberrypi3-poky-linux-gnueabi/u-boot/v2016.03+gitAUTOINC+df61a74e68-r0/git/board/raspberrypi/rpi_2/../rpi/rpi.c:340: undefined reference to `fw_dtb_pointer'
| /home/mickey/work/yocto/rpi-morty/build_u-boot/tmp/work/raspberrypi3-poky-linux-gnueabi/u-boot/v2016.03+gitAUTOINC+df61a74e68-r0/git/Makefile:1171: recipe for target 'u-boot' failed
| make[1]: *** [u-boot] Error 1
| make[1]: Leaving directory '/home/mickey/work/yocto/rpi-morty/build_u-boot/tmp/work/raspberrypi3-poky-linux-gnueabi/u-boot/v2016.03+gitAUTOINC+df61a74e68-r0/build'
```

It seems to missing symbol that fw_dtb_pointer.
I think fw_dtb_pointer is declared in lowlevel_init.S.
However lowlevel_init.S is not included in Makefiles.

I checked git log then I think suspicious a commit that 031e26954b95d49a66ea4677b9e867d08441f450.
This commit to avoid a failing build to u-boot_2016.11.
However morty's u-boot is 2016.03.

I reverted this commit and bitbake,then  build is succeeded.
Have you seen problems like this?

mirzak, 2017-01-24:
Yeah, that problem is expekted. 

That patch got applied before morty was created, and hence followed when morty branch was created. 

But I agree that it should be reverted on morty. It does still apply on master though.

mickey-happygolucky, 2017-01-24:
Thanks for your reply.
I understood.
I think so too that should not be reverted on master branch.

mirzak, 2017-01-24:
Sent a patch to yocto mailing list reverting this on morty...

agherzan, 2017-01-25:
Should be fixed now in master. Thank you.

closed 2017-01-25 by agherzan