agherzan/meta-raspberrypi #44: linux-raspberrypi_4.9.bb

mihaelaapetroaie, 2017-01-24:
[image]

agherzan, 2017-01-25:
Could you please replace the screenshot with text information about:
1. versions
2. host OS
3. full logs

Thank you,

agherzan, 2017-01-25:
In the meanwhile I compiled 4.9.X on master just fine. Please come back with more details. In the meanwhile I will consider this not reproducible.

closed 2017-01-25 by agherzan

mihaelaapetroaie, 2017-01-25:
added the info you asked for. Please let me know if you need anything else

Thanks

agherzan, 2017-01-25:
Can you use morty branches of poky and meta-raspberrypi? Those should be the stable branches to be used when not developing.

mihaelaapetroaie, 2017-01-25:
Thanks a lot for this. I used the meta-raspberrypi morty, but now I get a new error:

NOTE: Running task 720 of 2227 (/media/mika/new/poky/meta/recipes-bsp/u-boot/u-boot_2016.03.bb:do_fetch)
NOTE: recipe u-boot-v2016.03+gitAUTOINC+df61a74e68-r0: task do_fetch: Started
NOTE: recipe libgcc-6.2.0-r0: task do_install: Started
NOTE: recipe u-boot-v2016.03+gitAUTOINC+df61a74e68-r0: task do_fetch: Succeeded
NOTE: Running task 721 of 2227 (/media/mika/new/poky/meta/recipes-bsp/u-boot/u-boot_2016.03.bb:do_unpack)
NOTE: recipe u-boot-v2016.03+gitAUTOINC+df61a74e68-r0: task do_unpack: Started
NOTE: recipe u-boot-v2016.03+gitAUTOINC+df61a74e68-r0: task do_unpack: Succeeded
NOTE: Running task 722 of 2227 (/media/mika/new/poky/meta/recipes-bsp/u-boot/u-boot_2016.03.bb:do_patch)
NOTE: recipe u-boot-v2016.03+gitAUTOINC+df61a74e68-r0: task do_patch: Started
ERROR: u-boot-v2016.03+gitAUTOINC+df61a74e68-r0 do_patch: Command Error: 'quilt --quiltrc /media/mika/new/poky/build2/tmp/sysroots/x86_64-li$
Applying patch 0001-CONFIGS-rpi-enable-mender-requirements.patch
patching file include/configs/rpi.h
Hunk #1 FAILED at 118.
1 out of 1 hunk FAILED -- rejects in file include/configs/rpi.h
Patch 0001-CONFIGS-rpi-enable-mender-requirements.patch does not apply (enforce with -f)
ERROR: u-boot-v2016.03+gitAUTOINC+df61a74e68-r0 do_patch: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /media/mika/new/poky/build2/tmp/work/raspberrypi2-poky-linux-gnueabi/u-boot/v2016.03+gitAUTOINC+df61a74$
NOTE: recipe u-boot-v2016.03+gitAUTOINC+df61a74e68-r0: task do_patch: Failed
ERROR: Task (/media/mika/new/poky/meta/recipes-bsp/u-boot/u-boot_2016.03.bb:do_patch) failed with exit code '1'
NOTE: Running task 723 of 2227 (/media/mika/new/poky/meta/recipes-core/glibc/cross-localedef-native_2.24.bb:do_fetch)
NOTE: recipe make-4.2.1-r0: task do_fetch: Succeeded
NOTE: recipe cross-localedef-native-2.24-r0: task do_fetch: Started
NOTE: recipe cross-localedef-native-2.24-r0: task do_fetch: Succeeded
NOTE: recipe libgcc-6.2.0-r0: task do_install: Succeeded
NOTE: recipe linux-raspberrypi-1_4.4.43+gitAUTOINC+17ca1c9309-r0: task do_fetch: Succeeded
NOTE: Tasks Summary: Attempted 723 tasks of which 0 didn't need to be rerun and 1 failed.

agherzan, 2017-01-25:
You are applying a patch that is not part if meta-raspberrypi:

```
ERROR: u-boot-v2016.03+gitAUTOINC+df61a74e68-r0 do_patch: Command Error: 'quilt --quiltrc /media/mika/new/poky/build2/tmp/sysroots/x86_64-li$
Applying patch 0001-CONFIGS-rpi-enable-mender-requirements.patch
patching file include/configs/rpi.h
```

mirzak, 2017-01-25:
I assume you are are using meta-mender(morty), that error you get @mihaelaapetroaie got fixed today, so just do a pull and it should be ok.

Sorry for hijacking this issue, but I am involved in meta-mender as well so might as well :)

mihaelaapetroaie, 2017-01-26:
@mirzak, thanks a lot! It worked!!