agherzan/meta-raspberrypi #45: mouse does not worked at core-image-sato in morty

mickey-happygolucky, 2017-02-07:
Hello.

I tried to use core-image-sato with vc4graphics.
Sato is booted however mouse pointer does not moved in spite of mouse is moved.

I added the following line to my local.conf. 
It seems to work fine.

```txt
IMAGE_INSTALL_append = " xf86-input-evdev"
```

I think xf86-input-evdev should be included IMAGE_INSTALL or dependencies to core-image-sato or else...

Would you let me know your opinion.
Thanks.

agherzan, 2017-02-07:
The BSP is not maintaining that components and how the input driver is deployed. This issue should be directed to the yocto/poky/oe-core mailing list.

mickey-happygolucky, 2017-02-08:
mmm...

but other platform is not happend such a problem.
for exapmle meta-ti, xf86-input-evdev is included XSERVER varibale in omap4.inc.
Such as follows.

```txt
PREFERRED_PROVIDER_virtual/xserver = "xserver-xorg"
XSERVER = "xserver-xorg \
           xserver-xorg-module-exa \
           xf86-input-evdev \
           xf86-video-omap"
```

It really oe-core's issue?
Whether xf86-input-evdev should be used or not depends on the board?

I know already workaround. 
So I think that you do not have to fix meta - raspberypi by force.

Thanks.

agherzan, 2017-02-12:
We can address this in meta-raspberry too by adding this package to the XSERVER too. This will fix your problem so feel free to send a patch to mailing list.

mickey-happygolucky, 2017-02-12:
I should have send the patch to mailing list.
I am sorry, I do not know whether it was successfully transmitted.

mickey-happygolucky, 2017-02-16:
Discussion is ongoing in mailing list.
It seems that the problem was that meta-raspberrypi did not correspond to the following oe-core's commit.

```txt
xorg-xserver recommends xf86-input-libinput since oe-core's commit

| commit 2d005faff6341a81a2afae28860101ba9db51ae8
| Author: Jussi Kukkonen <jussi.kukkonen@intel.com>
| Date:   Wed Oct 26 11:37:38 2016 +0300
|
|    conf: Use xf86-input-libinput by default
| ...

```

I think it will be accepted Andreas Müller's patch.
I just want the mouse to work on core-image-sato, so that's fine.

Thanks.

Ixox, 2017-03-11:
I should have read more carefully the open issues earlier.
I also noticed that for any X11 built. Those lines have been removed in december 2016.
https://github.com/agherzan/meta-raspberrypi/commit/c40558173ffd96c499d101155f6c4c2be85d9f0f
They're present in several other BSP layer such as the beagleboard : 
https://github.com/beagleboard/meta-beagleboard/blob/master/common-bsp/conf/machine/beaglebone.conf

agherzan, 2017-03-21:
Current changes in master should get this back working. Can you give it a try?

closed 2017-03-21 by agherzan

mickey-happygolucky, 2017-04-10:
Sorry I'm late.
I tried in master. It works fine! thanks!!

By the way, would you fix it in morty branch?

agherzan, 2017-04-11:
@mickey-happygolucky Would you be able to cherry pick the needed patches from master and send a PR to morty? That would help.

mickey-happygolucky, 2017-04-13:
I tried cherry-pick the 52bec45b into morty. However mouse does not work.

I checked a log in a meta directory(oe-core).  I realized that a1668c8721709f3d1bc0078737514111e1985bdd is not in the oe-core in morty.
It seems 52bec45b depends on a1668c8721709f3d1bc0078737514111e1985bdd. 

I don't know what I should do...

I think easiest way to make mouse work in morty is revert c4055817.(It is workaround.)

Do you have any ideas?

agherzan, 2017-04-17:
Hi @mickey-happygolucky . I would take this to the mailing list but my recommendation would be to push a commit to morty which adds a bbeppend xserver-xorg which includes xf86-input-libinput in XSERVER_RRECOMMENDS.

mickey-happygolucky, 2017-04-21:
Hello @agherzan . Thanks for your reply. 

I have opened a PR according to your advice. Would you check it out?

agherzan, 2017-04-21:
Thanks @mickey-happygolucky . Looks good. Would I ask too much to push the commits to the mailing list? This is the official way of pushing commits for meta-raspberrypi as stated in README. In this way more people will have a chance of reviewing this change.

mickey-happygolucky, 2017-04-23:
Hello @agherzan . 

Okay I sent patches to yoctoproject mailing list.
Thanks.

PR #54 (2017-05-16): modify to use xf86-input-libinput
To fix #45 cherry-pick the 52bec45 and adds a bbeppend xserver-xorg which includes xf86-input-libinput in XSERVER_RRECOMMENDS.