agherzan/meta-raspberrypi #46: mouse does not worked at core-image-sato in morty

mickey-happygolucky, 2017-02-12:
I could not reopened follows issue, then I created new issue.
https://github.com/agherzan/meta-raspberrypi/issues/45

I said "I know already workaround. So I think that you do not have to fix meta-raspberypi by force." in comment.

BUT I would like to hear your opinion on the questions below.

```
but other platform is not happend such a problem.
for exapmle meta-ti, xf86-input-evdev is included XSERVER varibale in omap4.inc.
Such as follows.

PREFERRED_PROVIDER_virtual/xserver = "xserver-xorg"
XSERVER = "xserver-xorg \
           xserver-xorg-module-exa \
           xf86-input-evdev \
           xf86-video-omap"
It really oe-core's issue?
Whether xf86-input-evdev should be used or not depends on the board?
```

I do not necessarily need to fix this problem in meta-raspberrypi.
But I think you should answer this question.

agherzan, 2017-02-12:
Please don't create duplicated issue and wait patiently until me or somebody else has the time to give some feedback. If you can't reopen an issue, it will be opened by a maintainer when considered appropriate. Thanks for understanding.

closed 2017-02-12 by agherzan

mickey-happygolucky, 2017-02-12:
Sorry. 
Since the problem was closed too soon, I thought whether the question was ignored.