agherzan/meta-raspberrypi #47: Build braking for linux-raspberrypi_4.9.bb and linux-raspberrypi_4.4.bb

TheMeaningfulEngineer, 2017-02-13:
Hi, I'm getting this error when trying to run the build. Tried systems Lubuntu 16.10 and Lubuntu 16.04.
16.04 doesn't complain about being an unsupported distribution.

Here are the steps to reproduce the issue:

```
sudo apt-get install gawk wget git-core diffstat unzip texinfo gcc-multilib \
     build-essential chrpath socat libsdl1.2-dev xterm
git clone http://git.yoctoproject.org/git/poky
cd poky
git checkout morty
export POKY_DIR=$PWD
git clone http://git.yoctoproject.org/git/meta-raspberrypi
source oe-init-build-env rpi-build
sed -i "11 a\  $POKY_DIR/meta-raspberrypi \\\\" conf/bblayers.conf
sed -i 's/MACHINE ??= "qemux86"/MACHINE ??= "raspberrypi2"/' conf/local.conf
bitbake rpi-hwup-image
```

It results in the following error:

```
ERROR: /home/user/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.9.bb: Error executing a python function in <code>:                                                                                          | ETA:  --:--:--

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 18, function: <module>
     0014:__anon_18__home_user_poky_meta_recipes_kernel_linux_linux_yocto_inc(d)
     0015:__anon_364__home_user_poky_meta_classes_kernel_yocto_bbclass(d)
     0016:__anon_6__home_user_poky_meta_recipes_kernel_linux_linux_dtb_inc(d)
     0017:__anon_148__home_user_poky_meta_raspberrypi_recipes_kernel_linux_linux_rpi_inc(d)
 *** 0018:__anon_38__home_user_poky_meta_raspberrypi_recipes_kernel_linux_linux_raspberrypi_inc(d)
File: '/home/user/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi.inc', lineno: 36, function: __anon_38__home_user_poky_meta_raspberrypi_recipes_kernel_linux_linux_raspberrypi_inc
     0032:# Set programmatically some variables during recipe parsing
     0033:# See http://www.yoctoproject.org/docs/current/bitbake-user-manual/bitbake-user-manual.html#anonymous-python-functions
     0034:python __anonymous () {
     0035:    kerneltype = d.getVar('KERNEL_IMAGETYPE')
 *** 0036:    kerneldt = get_dts(d, d.getVar('LINUX_VERSION'))
     0037:    d.setVar("KERNEL_DEVICETREE", kerneldt)
     0038:}
     0039:
     0040:do_kernel_configme_prepend() {
Exception: TypeError: getVar() missing 1 required positional argument: 'expand'

ERROR: Failed to parse recipe: /home/user/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.9.bb
ERROR: /home/user/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.4.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 18, function: <module>
     0014:__anon_18__home_user_poky_meta_recipes_kernel_linux_linux_yocto_inc(d)
     0015:__anon_364__home_user_poky_meta_classes_kernel_yocto_bbclass(d)
     0016:__anon_6__home_user_poky_meta_recipes_kernel_linux_linux_dtb_inc(d)
     0017:__anon_148__home_user_poky_meta_raspberrypi_recipes_kernel_linux_linux_rpi_inc(d)
 *** 0018:__anon_38__home_user_poky_meta_raspberrypi_recipes_kernel_linux_linux_raspberrypi_inc(d)
File: '/home/user/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi.inc', lineno: 36, function: __anon_38__home_user_poky_meta_raspberrypi_recipes_kernel_linux_linux_raspberrypi_inc
     0032:# Set programmatically some variables during recipe parsing
     0033:# See http://www.yoctoproject.org/docs/current/bitbake-user-manual/bitbake-user-manual.html#anonymous-python-functions
     0034:python __anonymous () {
     0035:    kerneltype = d.getVar('KERNEL_IMAGETYPE')
 *** 0036:    kerneldt = get_dts(d, d.getVar('LINUX_VERSION'))
     0037:    d.setVar("KERNEL_DEVICETREE", kerneldt)
     0038:}
     0039:
     0040:do_kernel_configme_prepend() {
Exception: TypeError: getVar() missing 1 required positional argument: 'expand'
```

TheMeaningfulEngineer, 2017-02-13:
Expanding all `getVar('SOMETHING')` with `getVar('SOMETHING', False)` seems to solve the issue. 
Don't know the reasoning behind the fix :)

agherzan, 2017-02-13:
What version of poky and meta-rasperrypi are you using?

TheMeaningfulEngineer, 2017-02-14:
Poky git commit: `6a1f33cc40bfac33cf030fe41e1a8efd1e5fad6f`
meta-raspberrypi git commit: `68634deeed2a930a15eedd02e6b1de7d1e014d3b`

Will this do?
It's `morty` checked out today, and meta-raspberrypi also checked out today.

agherzan, 2017-02-14:
The problem is that you are using a master raspberrypi with morty poky. You should use morty with morty for both repositories. The issue you are facing is due to this commit:
3ad29d11e7f9c08de8e782ef4264e2047acb42bb .

TheMeaningfulEngineer, 2017-02-14:
I see. Thanks for the clarification. 
Will test this and close the issue on confirmation.

TheMeaningfulEngineer, 2017-02-14:
All seems to be in order now, thanks.
Here is a suggestion on the documentation change:

> 2. Yocto BSP Layer - RaspberryPi
> ================================
> 
> This layer depends on: 
> URI: git://git.yoctoproject.org/poky
> 
> NOTE: Make sure that you're using the branch that is compatible with your
> poky version. Example. For poky version morty, checkout the branch morty of the 
> meta-raspberrypi.

I found the whole openembedded dependency part somewhat confusing. And the note addresses the issue I had encountered.
Just my 2 cents :)

closed 2017-02-14 by TheMeaningfulEngineer