agherzan/meta-raspberrypi #48: question: cmdline.txt setup/usage

andreabenini, 2017-02-22:
Hi Andrei,
I'm using latest yocto (morty) synced with git, your layer and I'm using "raspberrypi" as MACHINE in order to keep compatibility with RPi1,2,3,0. I'm having issues with cmdline.txt because it seems to be rewritten by something else (maybe kernel?) and I want to set my own file. I've tried to create a .bbappend file with the do_deploy_append() function as reported [here](http://raspberrypi.stackexchange.com/questions/31842/how-do-i-generate-configure-config-txt-for-raspberry-pi2-using-yocto) but I don't know if my file is executed or if its output has been rewritten by something else. Do you have any hints or is there a reliable sample to look at ?

Thanks in advance

agherzan, 2017-02-22:
Hi Ben,

Indeed, Kernel is the only one who touches that file. Could you show me exactly what did you do in order to add you own modifications? What file did you create and where did you put it?

Andrei

andreabenini, 2017-02-23:
Here's something (meta-img/recipes-core/images/img-minimal.image.bb):
```
IMAGE_INSTALL += "dropbear shadow openssl gnupg ntpdate tzdata e2fsprogs e2fsprogs-resize2fs "

do_deploy() {
    echo "Do deploy configuration (cmdline.txt). moebius"
    install -d ${DEPLOYDIR}/bcm2835-bootfiles

    # Configuration file (config.txt)
    echo "" >>${DEPLOYDIR}/bcm2835-bootfiles/config.txt

    # Standard cmdline.txt output on production
    echo "dwc_otg.lpm_enable=0  console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes ipv6.disable=1 rootwait" > ${DEPLOYDIR}/bcm2835-bootfiles/output.txt
    # Standard cmdline.txt output on production
    echo "dwc_otg.lpm_enable=0  console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes ipv6.disable=1 rootwait" > ${DEPLOYDIR}/bcm2835-bootfiles/cmdline.txt
}
addtask do_deploy after do_install
```
MACHINE is "raspberrypi"
My file gets executed because other functions here are fine, this one seems to be ignored. I'm even trying to generate a dummy file `output.txt` but nothing appears on the boot partition or the deploy dir

agherzan, 2017-03-01:
The easiest way would be to go in tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/1_4.4.48+gitAUTOINC+7ddf96fbb7-r0/temp and check the run.do_deploy generated script.

andreabenini, 2017-03-01:
Fine, I'll look at it. Thanks.
general question: is it possible to execute a .bb file during the image creation without generating a "package" (.deb, rpm, pkg,...) inside your yocto image ? I'd like to add what you suggested in my image creation but without having a package (.deb) in my img.
Thanks again

closed 2017-03-04 by andreabenini