agherzan/meta-raspberrypi #49: devtool build linux-raspberrypi

gizero, 2017-02-28:
This was tested on current master. Only `poky/master` and `meta-raspberrypi/master` in `bblayers.conf`.

After starting a `devtool modify linux-raspberrypi` session, building the kernel fails with the following log. Any suggestion where to look for troubleshooting this?

```
MACHINE="raspberrypi3" devtool build linux-raspberrypi

Build Configuration:
BB_VERSION        = "1.33.1"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "raspberrypi3"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2+snapshot-20170228"
TUNE_FEATURES     = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU        = "hard"
meta              
meta-poky         
meta-yocto-bsp    = "master:3c83b56309ab419f8cda72c0711479f60f61439a"
meta-raspberrypi  = "master:c768a3d1aed8d16d08afe12fddb359914e0a203a"
workspace         = "<unknown>:<unknown>"

Initialising tasks: 100% |###########################################################################################################################################################################| Time: 0:00:00
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: linux-raspberrypi-1_4.4.48+git999-r0 do_configure: oe_runmake failed
ERROR: linux-raspberrypi-1_4.4.48+git999-r0 do_configure: Function failed: do_configure (log file is located at /scratch/gizero/poky-master-build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/1_4.4.4
8+git999-r0/temp/log.do_configure.4531)
ERROR: Logfile of failure stored in: /scratch/gizero/poky-master-build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/1_4.4.48+git999-r0/temp/log.do_configure.4531
Log data follows:
| DEBUG: Executing python function externalsrc_configure_prefunc
| DEBUG: Python function externalsrc_configure_prefunc finished
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: Executing python function check_oldest_kernel
| DEBUG: Python function check_oldest_kernel finished
| DEBUG: Executing shell function do_configure
| NOTE: make HOSTCC=gcc  HOSTCPP=gcc  -E oldconfig
| ERROR: oe_runmake failed
| make: *** No rule to make target 'oldconfig'.  Stop.
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /scratch/gizero/poky-master-build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/1_4.4.48+git999-r0/temp/log.do_configure.4531)
ERROR: Task (/home/gizero/work/upstreaming/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.4.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 452 tasks of which 128 didn't need to be rerun and 1 failed.
```

agherzan, 2017-03-01:
I personally don't know what's going on right now. I would say that it would be better to send an email to the mailing list. I personally don't usually mess with devtool.

agherzan, 2017-04-14:
User to send the question to the mailing list.

martingkelly, 2017-11-17:
I have also hit this issue. Was there ever a resolution to this?

martingkelly, 2017-11-18:
I think I have a solution to the issue; see referenced PR.

agherzan, 2017-11-18:
@martingkelly Nobody chased this one. Thanks for the fix. I'm currently on vacation for a week do you have this as a huge priority? I need to work a little to figure out what is going on. I'm not sure I understand from your PR exactly how dev tool build influences the creation of the config file.

martingkelly, 2017-11-18:
Hi, no problem waiting for a review.

I don't fully understand devtool; it appears that devtool just adds an inheritance from externalsrc.bbclass into the recipe. What I observed during debugging is that this line fails:

```
mv -f ${B}/.config ${B}/.config.patched
```

Because ${B}.config does not yet exist when we run this line, while in the normal build it does exist. It appears that ${B}.config normally gets created in kernel.bbclass via the lines I copied into `configure_setup`. This is why moving up the order and doing that logic in `do_configure_prepend` fixes the issue. That said, It's still a mystery to me why ${B}.config is getting created after our `do_configure_prepend` in the devtool/externalsrc case but not in the normal case.

I will try to figure out the root cause and let you know if I can. I know this patch appears to work but it makes me uncomfortable that I don't fully understand why.

PR #157 (2018-01-31): linux-raspberrypi: fix build with devtool
Currently, building linux-raspberrypi with "devtool build" breaks
because the do_configure_prepend function assumes that ${S} == ${B}.
Thus it tries to touch ${B}.config, but ${B}.config does not exist yet.
${S}.config. It will get created in the kernel_do_configure function in
kernel.bbclass, but that code hasn't happened yet.

Thus the correct workflow is:

(1) Copy ${WORKDIR}/defconfig to ${B}.config
(2) Tweak ${B}.config
(3) Run configure

Since we need to add (2) in between (1) and (3), and since (1) and (3)
are in kernel_do_configure (in kernel.b [... cut]
closed 2018-01-31 by agherzan