agherzan/meta-raspberrypi #51: rpi-sdimg file not found

BilalArain, 2017-04-06:
A quick question about sdimg file after building meta-raspberrypi. I have got rpi-hwup-image-raspberrypi-20170405083420.rootfs.tar.bz2 and bcm2835-bootfiles folder. I cannot see any file which says *sdimg*.

Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "raspberrypi"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2"
TUNE_FEATURES     = "arm armv6  vfp arm1176jzfs callconvention-hard"
TARGET_FPU        = "hard"
meta              
meta-poky         
meta-yocto-bsp    = "master:e016eb10b075e280b4e78a04e47b59a173386421"
meta-raspberrypi  = "master:da32aac453da278e254d37b816602410af85d162"

local.conf:
MACHINE ??= "raspberrypi"
DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_rpm"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
CONF_VERSION = "1"
IMAGE_FSTYPES = "tar.bz2 ext3.xz"

The image files in tmp/deploy/images/raspberrypi, after building successfully are as follows:
modules-1-4.9.13+git0+28ea32b9af-r0-raspberrypi-20170405083420.tgz
rpi-hwup-image-raspberrypi-20170405083420.rootfs.ext3.xz
rpi-hwup-image-raspberrypi-20170405083420.rootfs.tar.bz2
bcm2835-bootfiles (folder).

I am not sure which mailing-list I can post this question/issue, therefore please let me know if it is a known issue or if this may not be a suitable place to post this issue. Thanks.

roboticists, 2017-04-06:
I have found a problem and fixed it.

closed 2017-04-06 by roboticists

agherzan, 2017-04-06:
What was the issue?