agherzan/meta-raspberrypi #52: RPI multi board support ?

andreabenini, 2017-04-13:
Hi Andrei,
I'm able to compile a basic image (rpi-hwup-image.bb) for RaspberryPi (1/2/3) boards just by changing the MACHINE var, but when I set it for example to "raspberrypi" my image boots just only on v1 boards (B and B+), the same happens for "raspberrypi3" boards (3 only) and so on. Boot partition for each image is basically the same and has its kernel.img (or kernel7.img) only.
Is there a way to create an universal image (P1, P2, P3 support) that goes on each board just like what raspbian does ? I mean, I can compile with MACHINE set to "raspberrypi" and generate various kernels (1,2,3,0) to match the destination board, It's not heavily optimized but at least it runs on each single machine
On a cleaned up installation I have left busybox and kernel configs untouched because I'm doing just bare tests on them, each single image works but I wish to combine them. I might be able to compile each single kernel and with a dirty batch hack I can merge them, but, is there a Yocto way to do it ?

Thanks in advance for your reply
Ben

agherzan, 2017-04-14:
Hi @andreabenini . This should be brought in the mailing list as most of the users are there and will provide you additional information. I will reply there as soon as you send it. Sounds good?

andreabenini, 2017-04-14:
sounds reasonable, closing it

agherzan, 2017-04-14:
Thanks @andreabenini

closed 2017-04-14 by agherzan