agherzan/meta-raspberrypi #53: Waveshare 3.5 inches LCD

metamorphling, 2017-04-16:
Did anybody manage to make this display work with raspberry pi3?
In kernel config display drivers are being built as modules, but still no luck. LCD shows just white screen the moment I put power on. Tried to do some "echo 0 > /dev/fb0" and "tail -f /dev/input/touchscreen0", no reactions on touch too.

agherzan, 2017-04-17:
From this documentation it seems that it needs OOT drivers: http://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)#Driver . I don't think anyone worked in adding support for it.

agherzan, 2017-04-17:
Checked the archives there - probably you only need those overlays.

agherzan, 2017-05-12:
@metamorphling Were you able to make it work?

agherzan, 2017-05-24:
Hopefully you managed to make it work. It would be great if you give us some feedback. I'll close the ticket for now. Do comment if there is anything we can help with.

closed 2017-05-24 by agherzan