agherzan/meta-raspberrypi #56: Raspberry Pi 3 build issue

Tropicao, 2017-04-24:
Hello, I am encountering difficulties with the Rpi3 build. Since I was not sure the problem was not my fault, I cloned a fresh copy of poky and meta-raspberrypi (both checked out on morty, and meta dependencies on morty too), but the problem is still appearing, trying to build the DISTRO poky for MACHINE raspberrypi3 : 

➜  build git:(morty) bitbake rpi-hwup-image
Loading cache: 100% |#########################################################################################################################################################################################################| Time: 0:00:00
Loaded 2 entries from dependency cache.
WARNING: /home/alexis/Projets/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.9.bb: Exception during build_dependencies for PKG_kernel-image                                                                  | ETA:  --:--:--
WARNING: /home/alexis/Projets/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.9.bb: Error during finalise of /home/alexis/Projets/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.9.bb
WARNING: /home/alexis/Projets/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.4.bb: Exception during build_dependencies for PKG_kernel-image
WARNING: /home/alexis/Projets/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.4.bb: Error during finalise of /home/alexis/Projets/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.4.bb
ERROR: ExpansionError during parsing /home/alexis/Projets/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.9.bb
Traceback (most recent call last):
bb.data_smart.ExpansionError: Failure expanding variable PKG_kernel-image, expression was kernel-image-${@legitimize_package_name('${@get_kernelversion_headers('/home/alexis/Projets/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/1_4.9.21+git${SRCPV}-r0/linux-raspberrypi3-standard-build')}')} which triggered exception SyntaxError: invalid syntax (PKG_kernel-image, line 1)
Summary: There were 4 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

Unfortunately, since the problem seems to appear right at parsing, I cannot get the exact variable expansion with bitbake -e. Apparently my Yocto setup does not encounter the issue if I try the build for another RPI target, like raspberrypi0. Any advice on how to troubleshoot this issue ?

agherzan, 2017-04-24:
Please submit the issue to mailing list as there there more people who can help with feedback.

agherzan, 2017-05-17:
Tested it on morty and it seems working fine.

closed 2017-05-17 by agherzan