agherzan/meta-raspberrypi #57: raspberrypi3-64 fails to build

carlonluca, 2017-04-26:
Hello! I tried to build for raspberrypi2 and everything seemed to work properly. I then tried to build for raspberrypi3-64 but I got this error:

```
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 3 HOSTCC=gcc  HOSTCPP=gcc  -E Image CC=aarch64-poky-linux-gcc   -fuse-ld=bfd  --sysroot=/home/luca/yocto_mount/build_64/tmp/sysroots/raspberrypi3-64 LD=aarch64-poky-linux-ld.bfd    --sysroot=/home/luca/yocto_mount/build_64/tmp/sysroots/raspberrypi3-64 LOADADDR=0x00008000
| NOTE: make -j 3 HOSTCC=gcc  HOSTCPP=gcc  -E bcm2708-rpi-0-w.dtb
| ERROR: oe_runmake failed
|   CHK     include/config/kernel.release
|   GEN     ./Makefile
|   CHK     include/generated/uapi/linux/version.h
|   CHK     include/generated/utsrelease.h
|   Using /home/luca/yocto_mount/build_64/tmp/work-shared/raspberrypi3-64/kernel-source as source for kernel
|   CC      scripts/mod/empty.o
|   CC      scripts/mod/devicetable-offsets.s
|   GEN     scripts/mod/devicetable-offsets.h
|   MKELF   scripts/mod/elfconfig.h
|   HOSTCC  scripts/mod/modpost.o
|   HOSTCC  scripts/mod/file2alias.o
|   HOSTCC  scripts/mod/sumversion.o
|   HOSTLD  scripts/mod/modpost
|   CHK     include/generated/timeconst.h
|   CHK     include/generated/bounds.h
|   CHK     include/generated/asm-offsets.h
|   CALL    /home/luca/yocto_mount/build_64/tmp/work-shared/raspberrypi3-64/kernel-source/scripts/checksyscalls.sh
|   CHK     include/generated/compile.h
|   CC      scripts/mod/devicetable-offsets.s
|   CC      scripts/mod/empty.o
|   GEN     scripts/mod/devicetable-offsets.h
|   MKELF   scripts/mod/elfconfig.h
|   HOSTCC  scripts/mod/modpost.o
|   HOSTCC  scripts/mod/file2alias.o
|   HOSTCC  scripts/mod/sumversion.o
|   HOSTLD  scripts/mod/modpost
| make[3]: *** No rule to make target 'arch/arm64/boot/dts/bcm2708-rpi-0-w.dtb'.  Stop.
| arch/arm64/Makefile:114: recipe for target 'bcm2708-rpi-0-w.dtb' failed
| make[2]: *** [bcm2708-rpi-0-w.dtb] Error 2
| Makefile:150: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:24: recipe for target '__sub-make' failed
| make: *** [__sub-make] Error 2
| ERROR: Function failed: do_compile (log file is located at /home/luca/yocto_mount/build_64/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.9.23+gitAUTOINC+53460a0a50-r17/temp/log.do_compile.76508)
```

Any idea why this is happening?

betafive, 2017-05-12:
There was some discussion of this on the mailing list. @carlonluca, Is this now resolved?

carlonluca, 2017-05-14:
Yes, solved. It can be closed.

closed 2017-05-15 by betafive

bkzshabbaz, 2017-09-19:
How was this issue resolved?  I'm still seeing it on my end.