agherzan/meta-raspberrypi #58: Dependency rpi-basic-image, vc-graphics and integration xf86-video-fbturbo

denyste, 2017-04-27:
Hi Andrei,
I am creating and building an embedded Linux distribution for the raspberry pi3.
I use the version of Yocto Project 2.2.1 - Morty and the layer meta-raspberrypi.
I am integrating the vc-graphics/vc-graphics-hardfp (meta-raspberrypi recipes) into the image rpi-basic-image.
I am not able to solve the problems of dependencies between the recipe vc-graphics/vc-graphics-hardfp of this recipe rpi-basic-image (see the Error message below):

"1 error: Required build target 'rpi-basic-image' has no buildable providers.Missing or unbuildable dependency chain was: ['rpi-basic-image', 'vc-graphics']"

Could you help me to resolve the vc-graphics/vc-graphics-hardfp dependencies.

See also the copy of the IMAGE_INSTALL_append (Toaster Morty 2.2.1) below:
"apt dpkg dbus sudo tzdata glibc-utils localedef networkmanager pointercal xinit libmatchbox matchbox-terminal matchbox-desktop matchbox-keyboard xkeyboard-config matchbox-session matchbox-wm libx11 consolekit base-passwd liberation-fonts pkgconfig xserver-xorg xf86-input-libinput vc-graphics".

I also need xf86-video-fbturbo in the meta-raspberrypi.

Could you please help me to integrate the recipe for the driver xf86-video-fbturbo from https://github.com/ssvb/xf86-video-fbturbo into the meta-raspberrypi either yourself when it is easy for or tell me how to do it myself or if it is even possible.

Because the same driver applies in the Raspbian during the installation stage of the package xserver-xorg-video-fbturbo and I would like the same driver on my embedded Linux distribution. You would be my hero if you created the new recipe for raspberry pi3 for me.

See the links below : https://github.com/ssvb/xf86-video-fbturbo, https://github.com/ssvb/xf86-video-fbturbo/wiki/Installation.

https://www.reddit.com/r/raspberry_pi/comments/1ubcwu/fbturbo_accelerated_x_driver_in_raspbian/

I thank you in advance for your reply.

Denis

twoerner, 2017-04-27:
If you add the following to your local.conf file, your image will use vc4graphics instead of userland:

    MACHINE_FEATURES_append = " vc4graphics"

(note that the space after the opening double-quote and the 'v' is significant)

But I'm not 100% sure that will work for a morty build, you might have to move all your layers to master.

denyste, 2017-04-28:
Thanks Twoerner.
How I do this?
How do I move all layers morty branch to master?

betafive, 2017-05-12:
@denyste, just checkout the master branch in each layer repository.

twoerner, 2017-05-12:
Oops! I missed the followup question. Thanks @betafive!

agherzan, 2017-05-19:
@denyste How is it going?

agherzan, 2017-05-24:
Closing due to no activity.

closed 2017-05-24 by agherzan