agherzan/meta-raspberrypi #74: Readme is too long and needs breaking up

betafive, 2017-05-22:
Creating this so we don't forget...

Should we split the Readme into files in a doc directory in the git repository itself or should we use the wiki on GitHub? My preference is a doc directory as we're then not tied to GitHub too closely.

agherzan, 2017-05-22:
I would not go for github wiki either as I'm not sure how we can migrate in case we want to switch from it. I would vote though for something that can be interpreted / parsed / compiled in pdf and html.

betafive, 2017-05-22:
Ok. It sounds like a doc directory full of markdown formatted files is the way to go. We can then use sphinx or one of the thousand other similar tools to turn that into html/pdf.

PR #75 (2017-05-26): Split readme into docs directory and setup sphinx
* The top-level README document is split into multiple markdown files within a docs directory.

* Configuration for the Sphinx documentation system is created (using `sphinx-quickstart`). Documentation can now be built using `make html` or `make latexpdf` in the docs directory.

Fixes #74
closed 2017-05-26 by agherzan