agherzan/meta-raspberrypi #78: [morty]rpi-test-image: missing dependency

mirzak, 2017-05-26:
**Description**

**Steps to reproduce the issue:**

```
mirza@b58ada3e9601:~/project/raspberrypi/build$ bitbake rpi-test-image
WARNING: Host distribution "DebianGNULinux-8" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Parsing recipes: 100% |#############################################################################################################################################################################| Time: 0:00:23
Parsing of 1568 .bb files complete (0 cached, 1568 parsed). 2131 targets, 145 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'crda' (but /home/mirza/project/raspberrypi/layers/meta-raspberrypi/recipes-core/packagegroups/packagegroup-rpi-test.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'crda' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['crda']
NOTE: Runtime target 'packagegroup-rpi-test' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-rpi-test', 'crda']
ERROR: Required build target 'rpi-test-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['rpi-test-image', 'packagegroup-rpi-test', 'crda']

Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

The problem above is that I was following the instructions of the README of morty branch, and it does not mention dependencies on "meta-oe/meta-networking" and "meta-oe/meta-python". The `rpi-test-image` requires previously mentioned layers.

Creating this issue so that I wont forget to update the documentation. 

Question:

Shall we first convert README to markdown as we was done in master branch? And then update dependencies? Or the other way around?

betafive, 2017-05-26:
I think it's probably best to change this after merging #75 so that series doesn't need to be rebased.

agherzan, 2017-05-26:
@betafive I don't think we should bring the documentation work back in morty. It would be an additional work for which I fail to see amazing gain. I would say to let morty with old README, and only bring proper documentation in pyro an master. if you guys agree, we can have this addition only in morty for the README.

betafive, 2017-05-26:
Ah sorry I misread, if this is just for morty then yes just need to update the README.

agherzan, 2017-05-26:
Master already has this note in README. Thanks guys.

betafive, 2017-10-25:
Closing as the pull request above fixes this for morty.

commit 2a192261 (2017-09-19): README: update dependencies 

The rpi-test-image requires meta-networking and meta-python.

Fixes #78
closed 2017-10-25 by betafive