agherzan/meta-raspberrypi #85: raspberrypi3-64 MACHINEOVERRIDES

michal-mazurek, 2017-06-15:
Currently meta-raspberrypi/conf/machine/raspberrypi3-64.conf sets:
`MACHINEOVERRIDES = "raspberrypi3:raspberrypi:${MACHINE}"`
Is there a need for it to include "raspberrypi"? The non-64bit version doesn't seem to do that.

agherzan, 2017-06-15:
I think that is an issue. We should only have raspberrypi3 there and rely on `rpi` as family sock. @kraj , thoughts?

kraj, 2017-06-15:
I think it can be removed, if 32bit version doesnt use it then we dont need it for 64bit version either
we have to fix fallouts though.

agherzan, 2017-06-15:
@michal-mazurek Would you be able to sent a PR for this too?

betafive, 2017-10-25:
@michal-mazurek, @agherzan: Can this be closed now as the pull request above has been merged?

agherzan, 2017-10-27:
Yes. Let's clean. If there's anything left needed, I'll let people to comment. For now I'll close it.

closed 2017-10-27 by agherzan