agherzan/meta-raspberrypi #89: webkitgtk build is failed

markwang1971, 2017-06-27:
I have got the latest branch and its build is OK. But if webkitgtk is enabled, the build is failed because 
some interfaces are not defined in gl2ext.h. For example,  PFNGLFRAMEBUFFERTEXTURE2DMULTISAMPLEIMGPROC,  likely its version is too old.  Can I upgrade opengl es by myself?

agherzan, 2017-06-28:
Hello @markwang1971 . I'm not testing or playing with webkitgtk so I am not aware of this issue. What do you mean by updating opengl? I mean what provider?

markwang1971, 2017-06-28:
Hi agherzan, the default build of meta-raspberrypi uses userland to generate GLES2 header files in sysroot, which are too old and cause the failure of webkitgtk. 

But after adding MACHINE_FEATURE_append = " mesa-gl" (or mesa ), userland will be ignored and the mesa will generate GLES2 header files in sysroot, with which webkitgtk build can success.

agherzan, 2017-06-28:
I don't really understand why that MACHINE_FEATURE works with `mesa-gl`. The gles provider is set in meta-raspberrypi in rpi-default-providers.inc . Did you manually set mesa as gles provider?

markwang1971, 2017-06-29:
Sorry, it should be MACHINE_FEATURE_append = " vc4graphics", which causes userland to be replaced by mesa.

agherzan, 2017-06-29:
Most probably not. This sounds like and issue to be raised against the userland repository. They would be able to give us additional information.

kraj, 2017-09-07:
@markwang1971 yes update userland and also submit it upstream

betafive, 2017-10-25:
@markwang1971, any progress on raising this issue upstream?