agherzan/meta-raspberrypi #97: RPI CM3 64 Bits

Minipada, 2017-07-19:
**Description**
I am trying to write a configuration file for the cm3 64 bits, but I keep getting errors regarding `KERNEL_DEVICETREE`. I am trying to make the machine similar to the raspberrypi3-64 and add the `KERNEL_DEVICETREE` for the CM, but from what I understand, they are not shipped.
During compilation, it fails, not finding some dtb.It seem `KERNEL_DEVICETREE` points to some, but those are not generated. If not linked to `KERNEL_DEVICETREE` paramater, how can I ship the right dtb?
Here is my conf file:

    #@TYPE: Machine
    #@NAME: RaspberryPi Compute Module 3 (CM3-64)
    #@DESCRIPTION: Machine configuration for the RaspberryPi Compute Module 3 (CM3) in 64 bits mode

    MACHINEOVERRIDES = "raspberrypi3:${MACHINE}"

    require conf/machine/include/arm/arch-armv8.inc
    include conf/machine/include/rpi-base.inc

    KERNEL_DEVICETREE = " \
        broadcom/bcm2710-rpi-3-b.dtb \
        broadcom/bcm2837-rpi-3-b.dtb \
        broadcom/bcm2708-rpi-cm.dtb \
        broadcom/bcm2710-rpi-cm3.dtb \
        \
        overlays/hifiberry-amp.dtbo \
        overlays/hifiberry-dac.dtbo \
        overlays/hifiberry-dacplus.dtbo \
        overlays/hifiberry-digi.dtbo \
        overlays/i2c-rtc.dtbo \
        overlays/iqaudio-dac.dtbo \
        overlays/iqaudio-dacplus.dtbo \
        overlays/lirc-rpi.dtbo \
        overlays/pitft22.dtbo \
        overlays/pitft28-resistive.dtbo \
        overlays/pitft35-resistive.dtbo \
        overlays/pps-gpio.dtbo \
        overlays/rpi-ft5406.dtbo \
        overlays/w1-gpio.dtbo \
        overlays/w1-gpio-pullup.dtbo \
        overlays/pi3-disable-bt.dtbo \
        overlays/pi3-miniuart-bt.dtbo \
        overlays/vc4-kms-v3d.dtbo \
        "

    SERIAL_CONSOLE ?= "115200 ttyS0"
    VC4_CMA_SIZE ?= "cma-256"

    UBOOT_MACHINE = "rpi_3_config"
    MACHINE_FEATURES_append = " vc4graphics"

I am conscious this is not an issue itself, but once I understand how it works, I can make a PR here.

**Steps to reproduce the issue:**
1. Add the upper configuration file in `meta-raspberrypi/conf/machine`
2. Add `KERNEL_DEFCONFIG_raspberrypi-cm3-64 ?= "bcmrpi3_defconfig"` in `recipes-kernel/linux/linux-raspberrypi.inc`
3. `bitbake linux-raspberrypi`

**Describe the results you received:**

    |   DTC     arch/arm64/boot/dts/broadcom/bcm2837-rpi-3-b.dtb
    | make[3]: *** No rule to make target `arch/arm64/boot/dts/broadcom/bcm2708-rpi-cm.dtb'.  Stop.
    | make[2]: *** [broadcom/bcm2708-rpi-cm.dtb] Error 2
    | make[1]: *** [sub-make] Error 2
    | make: *** [__sub-make] Error 2

**Describe the results you expected:**
Compilation done with dtb from CM.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

betafive, 2017-07-27:
Looking in the linux-raspberrypi kernel sources, there isn't any `*-cm.dtb` files in https://github.com/raspberrypi/linux/tree/rpi-4.9.y/arch/arm64/boot/dts/broadcom.

I'd recommend asking on the linux-raspberrypi repository what the correct dts file is to for the RaspberryPi CM3 to be used in 64-bit mode.

Minipada, 2017-07-27:
You are right, that's where I mostly working. It's a WIP on the kernel and will update you asap. I'd like to merge it here once it's done

betafive, 2017-07-27:
Ok that's good news, it'd be nice to have the CM3 working with a 64-bit operating system.

Do you have a kernel branch that does have the dts files you need?

Minipada, 2017-07-27:
Yes, [here](https://github.com/synapticon/linux/tree/feature_rpi-cm3-64). For now, it is more experiments since it's the first time I work with this.

betafive, 2017-07-27:
Ok. For development you can modify `recipes-kernel/linux/linux-raspberrypi_4.9.bb` in the meta-raspberrypi layer, change SRC_URI to point to your kernel repository and branch. That should allow you to test your changes. Once they're working and have been merged in to the upstream linux-raspberrypi repository you can send us a pull request to add the raspberrypi-cm3-64 MACHINE configuration.

Minipada, 2017-07-27:
Yes, thanks for the informations. I'm really new to the kernel side, but am used to Yocto. This is pretty interesting, but I have to say, it lacks documentation, especially on some dts files. One needs to dig to understand why it's needed. e.g, why is the bcm2837 needed for the 64 bit rpi3.

Minipada, 2017-08-04:
Here as an update. I manage to boot, the kernel log seems really promising. My only issue is it seems device tree overlays are ignored, for a reason I ignore. Because they are ignored, I cannot use `pi3-disable-bt` enabling to log in. Hence the message `console [ttyS0] enabled` is not displayed

I think dtoverlays are ignored since adding overlay to use uart1 doesn't change any behavior.

The setup:
* rpi-basic-image
* [raspberrypi-cm3-64.conf](https://github.com/synapticon/meta-raspberrypi/tree/feature_rpi-cm3-64)
* [kernel 4.9.39](https://github.com/synapticon/linux/tree/feature_cm3_64). I also tested 4.8.16 and have same results
* I don't use `KERNEL_IMAGETYPE=uImage`. U-boot lacks the config file. I wrote it, but I got stuck trying to edit Makefiles in the kernel source to create the uImage. From what I read, there's a way to list device tree list in uboot. It would be helpful.
* cmdline.txt: `dwc_otg.lpm_enable=0 console=tty1 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait cma=128M@64M`
* config.txt: 
    * dtparam=spi=on
    * dtparam=i2c=on
    * dtparam=i2c_arm=on
    * enable_uart=1
    * dtoverlay=pi3-disable-bt
    * dtoverlay=vc4-kms-v3d
    * gpu_mem=16
    * device_tree=[bcm2837-rpi-cm3.dtb](https://github.com/synapticon/linux/blob/feature_cm3_64/arch/arm64/boot/dts/broadcom/bcm2837-rpi-cm3.dts) (WIP)
    * core_freq=250

Here is the complete log:

    [    0.000000] Booting Linux on physical CPU 0x0
    [    0.000000] Linux version 4.9.39 (dbensoussan@ip-whatever) (gcc version 5.4.0 (GCC) ) #1 SMP PREEMPT Thu Aug 3 15:28:23 UTC 2017
    [    0.000000] Boot CPU: AArch64 Processor [410fd034]
    [    0.000000] efi: Getting EFI parameters from FDT:
    [    0.000000] efi: UEFI not found.
    [    0.000000] cma: Reserved 128 MiB at 0x0000000004000000
    [    0.000000] percpu: Embedded 22 pages/cpu @fffffff9fdf8c000 s52504 r8192 d29416 u90112
    [    0.000000] Detected VIPT I-cache on CPU0
    [    0.000000] CPU features: enabling workaround for ARM erratum 845719
    [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 249984
    [    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ea00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait cma=128M@64M drm.debug=0x1f
    [    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
    [    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
    [    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
    [    0.000000] Memory: 852612K/1015808K available (6780K kernel code, 754K rwdata, 2548K rodata, 2688K init, 828K bss, 32124K reserved, 131072K cma-reserved)
    [    0.000000] Virtual kernel memory layout:
    [    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
    [    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
    [    0.000000]       .text : 0xffffff9a55480000 - 0xffffff9a55b20000   (  6784 KB)
    [    0.000000]     .rodata : 0xffffff9a55b20000 - 0xffffff9a55db0000   (  2624 KB)
    [    0.000000]       .init : 0xffffff9a55db0000 - 0xffffff9a56050000   (  2688 KB)
    [    0.000000]       .data : 0xffffff9a56050000 - 0xffffff9a5610ca00   (   755 KB)
    [    0.000000]        .bss : 0xffffff9a5610ca00 - 0xffffff9a561dba94   (   829 KB)
    [    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
    [    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
    [    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
    [    0.000000]               0xffffffbfe7000000 - 0xffffffbfe7f80000   (    15 MB actual)
    [    0.000000]     memory  : 0xfffffff9c0000000 - 0xfffffff9fe000000   (   992 MB)
    [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
    [    0.000000] Preemptible hierarchical RCU implementation.
    [    0.000000]  Build-time adjustment of leaf fanout to 64.
    [    0.000000] NR_IRQS:64 nr_irqs:64 0
    [    0.000000] Failed to get local register map. FIQ is disabled for cpus > 1
    [    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
    [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
    [    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
    [    0.000186] Console: colour dummy device 80x25
    [    0.001397] console [tty1] enabled
    [    0.001443] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=19200)
    [    0.001507] pid_max: default: 32768 minimum: 301
    [    0.001842] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
    [    0.001883] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
    [    0.002833] Disabling cpuset control group subsystem
    [    0.003018] ftrace: allocating 23498 entries in 92 pages
    [    0.064943] ASID allocator initialised with 65536 entries
    [    0.072885] EFI services will not be available.
    [    0.081879] Detected VIPT I-cache on CPU1
[... 126 lines cut ...]
    [    2.701010] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[... cut ...]
    ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQCvOlVjfx+9TKwtsIbWJ8fBcdyt+4aqNtAEFQWkKu6Ccz62N44xgO7Ha5hNZ/btUOPM6ydCPwhAo9Eg4UlUPFRkJ9XkcCiNQ0Dv2FTMMQWgvcsSOmSY1RO4Ho1GUY36IWvW2lcKJ7JT5zopjBncHdj0QPWMqUXw7YoLAOsn4GFRvR3D7yAMs+Kjv6cObOPonlwpuELYhG2nhHSJBD65V6u1zj5mwAJ48V8RPAEuUqLobKyw2rsVUNm7QN8eK5xDBsmbG6Ed94DxnzVPE7g16SjHtx+M8EjImPQoyckQI+IAEg7A+TM0VLfuM7P00a6U8Pxati3HK2MKE9nc53BzBgGl root@raspberrypi-cm3-64
    Fingerprint: md5 8f:6f:2c:1c:f1:8b:9e:dd:72:42:d5:5a:4c:b8:b1:62
    [    9.269227] NET: Registered protocol family 10
    dropbear.
    Starting syslogd/klogd: done
    INIT: Id "S0" respawning too fast: disabled for 5 minutes

agherzan, 2017-08-23:
Why don't you use the default serial console? And figure it out from there.

Minipada, 2017-08-23:
Hey @agherzan, I worked on it since last update, but I was not working for almost 3 weeks or so. I'll update the status in the next week

Minipada, 2017-09-05:
I am almost done but I still cannot make have the HDMI working. I found out vc4 driver was not loaded. I then tried with a raspberrypi3-64 (not compute module) machine but no success, I have exactly the same issue, videocore is not loaded.I'm using the minimum layers for creating the machine and here is my setup. 

* Config.txt:

      #Enable VC4 Graphics
      dtoverlay=vc4-kms-v3d
      gpu_mem=16
      device_tree=bcm2710-rpi-3-b.dtb
      core_freq=500
      sdram_freq=600
      arm_freq=1150
      enable_uart=1

* cmdline.txt:
      net.ifnames=0 dwc_otg.lpm_enable=0 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait cma=8M

* This is my local.conf:

      MACHINE ?= "raspberrypi3-64"
      DISTRO ?= "poky"

      EXTRA_IMAGE_FEATURES = " debug-tweaks package-management tools-profile "
      BB_NUMBER_THREADS ?= "16"
      PARALLEL_MAKE ?= "-j 16"

* I am using meta-raspberrypi@8527a396d38aec7776182caca8b9a237e03c3965 rpi-basic-image

This should work. I should have the videocore loaded at least during boottime, which I don't. Is there anything I'm missing?

betafive, 2017-09-21:
@Minipada - HDMI output not working is a known issue with raspberrypi3-64, it's now in the documentation. We're looking at get it fixed though. Khem's latest patches may help here and these have been merged to master - could you re-try with the latest master? If that doesn't work, I've got another possible solution in progress as well which we could try.

Minipada, 2017-09-21:
@betafive Thanks for the hints, I'll try after the current tests. I managed to use the screen using the fkms, I get a warning but this is enough to debug the dts properly. I have made some progress after understanding the device tree wiki and answers from the pelwell, the rpi DT master.
Thanks to the screen (I still don't have ttyS0 access), I see the firmware loads by default the device tree starting with the trailer `bcm27`, so I have modified my config.txt to force the new bcm2837.

I discovered it today, some parts of the `bcm27` dts are not loaded on the CM3 - 64 bits. I saw this reading the device tree in /proc/device-tree. 

I will push my changes tomorrow morning and will post here where to get them.

Minipada, 2017-10-04:
Please excuse me @betafive for the very late answer. [Here](https://github.com/synapticon/linux/tree/cm3_64) is the WIP with a description of the current status.

betafive, 2017-10-25:
@Minipada, are you planning to upstream this to the linux-raspberrypi kernel or carry this as a patch in meta-raspberrypi? HDMI output in 64-bit mode should definitely be working now so it would be good to get support for raspberrypi-cm3-64 as a MACHINE if possible.

Minipada, 2017-10-25:
Hello @betafive I have actually made it work and updated my PR on the raspberryi/linux repository but got no answer. I am going to push the tests in some days, the OS will be used by the team and I will have more assurance of the quality of the work done
Here is the link: https://github.com/raspberrypi/linux/pull/2147

betafive, 2017-10-25:
Ok, let's see how it goes upstream. Give them a reminder if needed.

agherzan, 2017-10-27:
pelwell is usually quite responsive so I think he will reply soon.

betafive, 2018-02-26:
@Minipada, looks like upstream haven't responded :( Could you rebase and try submitting it again? If upstream is still unresponsive we could carry the patches here for now.

kraj, 2018-04-16:
@Minipada is this still needed ? if so then the upstream pull branch needs refresh

agherzan, 2018-05-09:
@Minipada Friendly ping on this issue.

Minipada, 2018-05-09:
@agherzan thanks and sorry I didn't answer. I'll take a look at it tomorrow

Minipada, 2018-05-11:
The PR on the kernel repo looks more like a draft, I got requested some changes and am working on them and doing some cleanup.

agherzan, 2018-07-26:
@Minipada More people are asking about this. No pressure :)

Minipada, 2018-07-27:
I updated my pull request on the linux repository and have a PR ready for sumo as soon as they review and merge it (code [here](https://github.com/synapticon/meta-raspberrypi/tree/sumo-rpi-cm3-64))

agherzan, 2018-07-28:
Cheers @Minipanda.

Minipada, 2018-08-14:
My PR is still without answer since some time, but I just saw this got merged https://github.com/raspberrypi/linux/commit/248e0f711ec517aff69b815bac7b884bd3a04679

Is it hope?

stevep-korg, 2018-08-16:
Issue 304 is related to this issue.

https://github.com/agherzan/meta-raspberrypi/issues/304

As you have discovered,  the dts file now exists in the raspberrypi linux repo, and the Makefile has also been fixed up.

Once the meta-raspberry recipes can leverage this change to raspberrypi linux, then the meta-raspberrypi kernel device tree could be updated to include the cm3.

stevep-korg, 2018-09-15:
For now, if you wish to have a functional CM3 dtb built and deployed to the boot folder for your raspberrypi3-64 target, simply add the following to your local.conf:

KERNEL_DEVICETREE_append = "broadcom/bcm2710-rpi-cm3.dtb"

It's working well for me at head of sumo, where the linux-raspberry pi kernel recipe fetches a recent enough version where the cm3 dts is present, so the build of the dtb succeeds now.

Minipada, 2018-09-15:
I'll try asap, and close this.

stevep-korg, 2018-09-15:
Note, I am building the 4.14 flavor in meta-raspberrypi (4.14.68), sumo branch, and that worked.

I have not built for a 4.9 flavor in some time.

Minipada, 2018-09-17:
@stevep-korg I couldn't boot the image.
This is my local.conf:

```
MACHINE = "raspberrypi3-64"

EXTRA_IMAGE_FEATURES = " \
  allow-empty-password \
  debug-tweaks \
  empty-root-password \
  package-management \
  ssh-server-openssh \
"

BB_NUMBER_THREADS = "32"
PARALLEL_MAKE = "-j 32"

DL_DIR = "/home/build/yocto/yocto_topdir/downloads"
SSTATE_DIR = "/home/build/yocto/yocto_topdir/sstate-cache"
TMPDIR = "/home/build/yocto/yocto_topdir/tmp"

KERNEL_DEVICETREE_append = "broadcom/bcm2710-rpi-cm3.dtb"
```

The module is there. I have nothing either on the hdmi, nor on serial. I tried using rpi-basic-image and core-image-base (as I noticed the former indicated it was deprecated)

My layers:
* poky master
* meta-openembedded master
* meta-raspberrypi sumo

stevep-korg, 2018-09-17:
OK great, glad you could boot it.

In terms of the UART communication stopping, we have not seen that.  We use the UART for MIDI.  We set the following in config.txt

# configure uart0 for serial midi (/dev/ttyAMA0)
enable_uart=1
# midi-uart0: fakes out tty driver so that baudrate 38400 yields 31250
dtoverlay=midi-uart0

BTW, I also put all repos in branch sumo that can be in sumo.  I find master to be too unstable for regular development purposes in my organization.

I build rpi-hwup-image.   I don't care right now that it's deprecated; it has what I need for now  My organization will probably update our recipes to continue to support the rpi-hwup-image. target, even if it is retired from meta-raspberry pi.  Also, beware of core-image-minimal if you need kernel modules; it doesn't package them into the image!

Minipada, 2018-09-24:
I confirm that everything works perfectly. Thanks @stevep-korg for the simple and efficient patch. I'm now closing it

closed 2018-09-24 by Minipada