agherzan/meta-raspberrypi #101: Provider conflicts

yangjiel, 2017-07-26:
When I try to build all the packages in morty branch, I have couple errors:

ERROR: Nothing PROVIDES 'virtual/libgles1' (but /home/gumstix/Buildbot/yocto-raspberrypi3-morty/build/poky/meta/recipes-graphics/mesa/mesa-demos_8.3.0.bb DEPENDS on or otherwise requires it)
ERROR: mesa PROVIDES virtual/libgles1 but was skipped: PREFERRED_PROVIDER_virtual/libgl set to mesa-gl, not mesa
ERROR: mesa PROVIDES virtual/libgles1 but was skipped: PREFERRED_PROVIDER_virtual/libgl set to mesa-gl, not mesa
ERROR: ti-sgx-ddk-um PROVIDES virtual/libgles1 but was skipped: PREFERRED_PROVIDER_virtual/egl set to userland, not ti-sgx-ddk-um

I fixed this by changing the rpi-default-providers.inc of virtual/egl, virtual/libgles2, virtual/libgl, virtual/mesa to mesa, then I have other errors:

ERROR: Nothing PROVIDES 'userland' (but /home/gumstix/Buildbot/yocto-raspberrypi_cm3-morty/build/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.8.3.bb DEPENDS on or otherwise requires it)
ERROR: userland was skipped: PREFERRED_PROVIDER_virtual/libgles2 set to mesa, not userland

I fixed this by changing the gstreamer1.0-plugins-bad_%.bbappend 
-PACKAGECONFIG[dispmanx] = "--enable-dispmanx,--disable-dispmanx,userland"
+PACKAGECONFIG[dispmanx] = "--enable-dispmanx,--disable-dispmanx,virtual/egl virtual/libgles2"
and there seems no error.

Could you fix those problem upstream?

**Steps to reproduce the issue:**
1. Clone meta-openembedded, meta-raspberrypi, poky, meta-qt5 in morty branch
2. $ bitbake mesa-demos
3. $ bitbake gstreamer1.0-plugins-bad

betafive, 2017-07-27:
Could you try reverting your changes and setting the following in your local.conf file:

    MACHINE_FEATURES += "vc4graphics"

This should then set the correct providers for virtual/egl, virtual/libgles2, etc and remove "dispmanx" from the PACKAGECONFIG for gstreamer1.0-plugins-bad.

We may need to improve the documentation to say which packages require vc4graphics to be enabled.

yangjiel, 2017-07-27:
Thank you for your reply and helping, there is no such error anymore.

closed 2017-07-27 by yangjiel