agherzan/meta-raspberrypi #106: Support for multiple displays

d21d3q, 2017-08-06:
There is currently one Waveshare display supported ([here](https://github.com/agherzan/meta-raspberrypi/blob/f6a2ca21c72b8d97cd0f89a0a436bf90b431698b/recipes-bsp/bootfiles/rpi-config_git.bb#L121)).
It is being done by setting variable `WAVESHARE_1024X600_C_2_1`
```bash
if [ "${WAVESHARE_1024X600_C_2_1}" = "1" ]; then
# set some configuraton.
fi
```
I am going to add support for different displays (which have different parameters). I am thinking about doing it in way like this: 
```bash
if [ "${HDMI_DISPLAY_TYPE}" = "WAVESHARE_7_INCH_LCD_C" ] || [ "${WAVESHARE_1024X600_C_2_1}" = "1" ]; then 
# config
elif [ "${HDMI_DISPLAY_TYPE}" = "WAVESHARE_10_1_INCH_LCD_B" ]; then 
# config
elif [ "${HDMI_DISPLAY_TYPE}" = "WAVESHARE_10_1_INCH_LCD_B_ALTERNATE" ]; then
# config
fi
```
First line is for backward compatibility. Is that needed?
According to naming convention I would stick with naming from their [wiki](http://www.waveshare.com/wiki/Main_Page) which contains configuration parameters.
On the other hand, those configs differ mostly with `hdmi_cvt` parameter which corresponds to resolution, and sometimes with `hdmi_mode`. I am playing around with [this one](http://www.waveshare.com/wiki/10.1inch_HDMI_LCD_(B)_(with_case)) and i had to use alternate configuration.
Maybe we should stick with resolution based naming, and some other variable for alternate `hdmi_mode`, or one value per configuration with `_ALTERNATE` suffix? or `_MODE_27`, '_MODE_87` 

BTW, I've just found out that in rpi [documentation](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md) there is nothing about `hdmi_mode = 87`

twoerner, 2017-08-09:
> First line is for backward compatibility. Is that needed?

I'm most likely the only person using that option (I was the one who added it). If we're going to change to a new style of specifying the panel (i.e. from resolution to product description) then we shouldn't keep the older style for backwards compatibility.

It's no problem for me to change it in my config files. Over time this sort of backwards compatibility ends up looking silly :-)

betafive, 2017-10-25:
This sounds like a good enhancement, any progress on this @d21d3q ?