agherzan/meta-raspberrypi #107: WiFi driver error

fwag, 2017-08-06:
**WiFi driver init fail**

I'm using the pyro version of this Yocto layer and when I boot up the image I receive the following error:
[    3.392744] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2

This is my bblayers variable:
BBLAYERS ?= " \
  /opt/yocto/poky/meta \
  /opt/yocto/poky/meta-poky \
  /opt/yocto/poky/meta-yocto-bsp \
  /opt/yocto/poky/meta-openembedded/meta-oe \
  /opt/yocto/poky/meta-openembedded/meta-multimedia \
  /opt/yocto/poky/meta-openembedded/meta-networking \
  /opt/yocto/poky/meta-openembedded/meta-python \
  /opt/yocto/poky/meta-raspberrypi \
"

This is my local.conf
MACHINE ?= "raspberrypi3"
DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_ipk"
SDKMACHINE ?= "x86_64"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
CONF_VERSION = "1"
PREFERRED_PROVIDER_udev = "eudev"
VIRTUAL-RUNTIME_init_manager = "sysvinit"
DISTRO_FEATURES_remove = "x11 wayland"
MACHINE_FEATURES_remove = "apm"
ENABLE_UART = "1"
ENABLE_RPI3_SERIAL_CONSOLE = "1"
RM_OLD_IMAGE = "1"
INHERIT += "rm_work"
LICENSE_FLAGS_WHITELIST = "commercial"
INHERIT += "extrausers"
EXTRA_USERS_PARAMS = "usermod -P changeme root; "
WIFI_SUPPORT = " \
    crda \
    iw \
    linux-firmware-bcm43430 \
    wireless-tools \
    wpa-supplicant \
"
IMAGE_INSTALL += "${WIFI_SUPPORT}"

I've used the command:
bitbake rpi-basic-image

Thanks for your support

agherzan, 2017-08-23:
@betafive Do you have ny idea if pyro has broken wifi?

embexus, 2017-09-27:
Pyro has no problem with wifi just built one image now, wifi working correctly !

agherzan, 2017-09-30:
So we can close this ticket?

fwag, 2017-09-30:
Can you share the configuration files ?

agherzan, 2017-09-30:
@fwag are you using pyro branch for both poky and meta-raspberrypi?

fwag, 2017-09-30:
yes of course

fwag, 2017-10-06:
if it could help I have seen the following warning while compiling linux-firmware:
WARNING: linux-firmware-1_0.0+gitAUTOINC+b14134583c-r0 do_install: linux-firmware stopped providing brcmfmac43430 v7.45.41.26.

fwag, 2017-10-13:
I finally solved it removing meta-rpi and just using meta-raspberrypi (bblayers.conf).
I have added in my local.conf the following lines:
DISTRO_FEATURES += "wifi"

IMAGE_INSTALL_append = " \
    crda \
    iw \
    linux-firmware-bcm43430 \
    linux-firmware-ralink \
    linux-firmware-rtl8192ce \
    linux-firmware-rtl8192cu \
    linux-firmware-rtl8192su \
    wireless-tools \
    wpa-supplicant \
    dhcp-client \
"
Thanks

closed 2017-10-13 by fwag