agherzan/meta-raspberrypi #108: set_image_size total number of sectors 81920 not a multiple of sectors per track 63  mtools_skip_check=1 to .mtoolsrc /pi/yocto4/poky/meta-rasp/re-c/ima/rpi-basic-image.bb

ehrenmann1977, 2017-08-15:
**Description**

**Steps to reproduce the issue:**
1. Yocto-Morthy
2. meta-raspberrypi
3. meta-diet
4. enable the systemd in local.conf

**Describe the results you received:**
failed due to image size

betafive, 2017-08-16:
Please could you send us the full bitbake output including the error.

agherzan, 2017-08-23:
@ehrenmann1977 were you able to find the roots of the problem? We would like to find out if so. Otherwise we will be closing this issue.

ehrenmann1977, 2017-08-24:
hi there,

yes i could solve this issue, i created a file at ~/.mtoosrc

Add mtools_skip_check=1 to skip this test

jku, 2017-09-07:
Are you 100% sure this is on morty or newer? I'm fairly confident I've fixed this before morty release in oe-core (commit "mtools: Patch out a useless sanity check").

betafive, 2017-09-21:
Closing this as there's a workaround in place and the issue should be fixed in oe-core as per comment above from @jku. Feel free to re-open if needed.

closed 2017-09-21 by betafive