agherzan/meta-raspberrypi #111: Build fails

agherzan, 2017-08-23:
**Description**

The build fails with current poky master.

**Steps to reproduce the issue:**
1. poky master
2. build rpi-test-image

**Describe the results you received:**

```
ERROR: No recipes available for:
  /work/yocto/raspberrypi/meta-raspberrypi/recipes-multimedia/gstreamer/gstreamer1.0-omx_1.10%.bbappend
```

**Describe the results you expected:**

Successful build.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

agherzan, 2017-08-23:
@kraj I'm going to try to port the patches to 1.12.

kraj, 2017-08-23:
I had patch on my fork already just proposed it for master

agherzan, 2017-08-23:
I thought so. OK. I'll let the builds finish and merge it. Thanks.

kraj, 2017-09-07:
this should be closed

closed 2017-09-11 by agherzan