agherzan/meta-raspberrypi #122: Wrong version gstreamer1.0-omx_1.10%.bbappend in pyro branch

wian007, 2017-09-12:
**Description**
recipes-multimedia/gstreamer/gstreamer1.0-omx_1.10%.bbappend  generates the error 
"ERROR: No recipes available for:" on the pyro branch.

Poky pyro switched to:
meta/recipes-multimedia/gstreamer/gstreamer1.0-omx_1.12.2.bb

This master branch has the correct bbappend.

**Steps to reproduce the issue:**
1. Clone the poky pyro branch
2. Clone the meta-raspberrypi pyro branch
3. Bitbake

wian007, 2017-09-12:
Suggested fix, merge master to pyro for gstreamer1.0* recipes

betafive, 2017-09-13:
Looks like it's still 1.10.x on pyro: http://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-multimedia/gstreamer?h=pyro

Are you sure you have the correct Poky branch checked out?

wian007, 2017-09-13:
I'm so sorry, though I double checked it but it seems I have the wrong branch checked out.
Guess I need more sleep

closed 2017-09-13 by wian007

agherzan, 2017-09-14:
Don't worry about it @wian007 . Glad you found the issue.