agherzan/meta-raspberrypi #124: RPi WIC image build does not include bootfiles

embexus, 2017-09-14:
wenn building Rpi image using WIC format the image does not include bootfiles and hence is not able to boot !

embexus, 2017-09-14:
following patch fix the issue:

diff --git a/conf/machine/include/rpi-base.inc b/conf/machine/include/rpi-base.inc
index 4a0ea2a..fe4849b 100644
--- a/conf/machine/include/rpi-base.inc
+++ b/conf/machine/include/rpi-base.inc
@@ -48,7 +48,7 @@ MACHINE_FEATURES += "apm usbhost keyboard vfat ext2 screen touchscreen alsa blue
 # Raspberry Pi has no hardware clock
 MACHINE_FEATURES_BACKFILL_CONSIDERED = "rtc"
 
-MACHINE_EXTRA_RRECOMMENDS += " kernel-modules"
+MACHINE_EXTRA_RRECOMMENDS += " kernel-modules bcm2835-bootfiles"
 
 # Set Raspberrypi splash image
 SPLASH = "psplash-raspberrypi"

embexus, 2017-09-14:
Adding MACHINE_EXTRA_RRECOMMENDS += " kernel-modules bcm2835-bootfiles rpi-config" in conf/machine/include/rpi-base.inc should fix this issue

gizero, 2017-09-14:
I use to build raspberrypi3 images generated by both `wic` and `sdcard_image_rpi` class, but still only use the latter for production. However, I'd bet the `wic` one used to boot fine at least until a few weeks ago. I remember I compared the vfat partitions content from the two images and they looked identical in the end. I'll get back to these observations and see if things have changed.

embexus, 2017-09-14:
This issue has been identified on pyro branch, didn't test master

agherzan, 2017-09-15:
The problem is that when using wic and not generating the sdcard, nobody depends on bcm2835-bootfiles and those files will not be deployed. Most probably you either generated both - sdcard first, or you had it build already.

gizero, 2017-09-15:
Indeed `wic` was added via `IMAGE_FSTYPES_append = " wic"`, always alongside with its default value from machine config so it's very likely what you say. If different `IMAGE_FSTYPES` get processed in order, I definitely never tried `wic` "alone".

betafive, 2017-09-15:
I have tested pull request https://github.com/agherzan/meta-raspberrypi/pull/120 with `IMAGE_FSTYPES = "wic"` for both raspberrypi3 and raspberrypi3-64. That does pick up the correct files now and the images do boot successfully.

I'm pretty sure the change should be to `IMAGE_BOOT_FILES` not `MACHINE_EXTRA_RRECOMMENDS` as the files need to go in the boot partition rather than the rootfs. I may be wrong there though I've not looked at the effect of adding them to `MACHINE_EXTRA_RRECOMMENDS`.

The dependencies are set by extending `do_image_wic[depends]` in that pull request.

agherzan, 2017-09-15:
That indeed is the right fix. We do need though to backport it in pyro.

betafive, 2017-09-15:
Agreed. I'm away for the next couple of days (moving house & office) but after that I'm going to take another look at pyro backports.

agherzan, 2017-09-15:
Thanks @betafive

agherzan, 2017-09-25:
This should be fixed in master. Ping us if that is not the case.

closed 2017-09-25 by agherzan