agherzan/meta-raspberrypi #131: WIC Image / u-Boot

embexus, 2017-09-27:
Version Pyro:
In case I build a WIC image and select my KERNEL_IMAGETYPE = "uImage" and add IMAGE_BOOT_FILES += "u-boot.bin  boot.scr"
The Rpi dos not boot up. To fix the problem :
I need to rename manually  kernel.img to uImage and u-boot.bin to kernel.img.
I have no idea how to perform this operation automatically with WIC (like in sdcard_image-rpi.bblcass) so that WIC image is correctly generated, Any Tipps ?

Thank you !

embexus, 2017-09-27:
Ok I figured it out in case of KERNEL_IMAGETYPE = "uImage" , IMAGE_BOOT_FILES  should be set to:

IMAGE_BOOT_FILES ?= "bcm2835-bootfiles/* \
                 ${@make_dtb_boot_files(d)} \
	         u-boot.bin;${SDIMG_KERNELIMAGE} \
		 boot.scr \
                 ${KERNEL_IMAGETYPE} \
                 "

betafive, 2017-09-27:
I know this is working on the master branch as I tested it with recent changes. We probably need to look at backporting 04a33ead and maybe 43e0169a and c26e9d3d from master to pyro.

I've been planning on looking at pyro backports for a while but I've been delayed due to my main internet connection being down. I can't make any promises on when I'll get to this, but I will look at it as soon as I get chance.

agherzan, 2017-10-20:
@embexus this should be fixed now in pyro. Can you confirm?

embexus, 2017-10-20:
yes it's fixed by the commit :40447de4782d76f1e23e67ba05e272c27f6ec250

closed 2017-10-20 by agherzan