agherzan/meta-raspberrypi #132: weston dependencies not resolved when using userland

fgiancane8, 2017-09-29:
Hi,
I am having a discussion in the oe-core mailing list, which I report here.
http://lists.openembedded.org/pipermail/openembedded-core/2017-September/142897.html

I am having issues in building weston (and Xwayland) when the provider for the openGL stack is selected to be "userland".

In the discussion, it is pointed out that "userland" recipe should RPROVIDES also the libegl, which actually does not.

I thought it was an error in the xorg-xserver recipe, but they are arguing that it is the BSP layer that should take care of fixing this.

Would you mind double-checking this issue with me? Thanks

fgiancane8, 2017-09-29:
From 1db90106c25bcc9fae548db5e123bf05bd69946d Mon Sep 17 00:00:00 2001
From: Francesco Giancane <francescogiancane8@gmail.com>
Date: Fri, 29 Sep 2017 16:48:44 +0200
Subject: [PATCH] userland: RPROVIDES is missing libegl

userland package is the Raspberry Pi 3 provider for the openGL stack.
If selected, it shall provide the gles2 and egl stacks in conjunction
with mesa-gl.

libegl was missing in the RPROVIDES variable, thus some run-time
dependencies were not met when using userland as provider.

---
 recipes-graphics/userland/userland_git.bb | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/recipes-graphics/userland/userland_git.bb b/recipes-graphics/userland/userland_git.bb
index 1fc0717..d4624bf 100644
--- a/recipes-graphics/userland/userland_git.bb
+++ b/recipes-graphics/userland/userland_git.bb
@@ -10,7 +10,7 @@ PR = "r5"
 PROVIDES = "virtual/libgles2 \
             virtual/egl"
 
-RPROVIDES_${PN} += "libgles2 egl"
+RPROVIDES_${PN} += "libgles2 egl libegl"
 
 COMPATIBLE_MACHINE = "^rpi$"
 
-- 
2.13.5

fgiancane8, 2017-09-29:
possible patch here...

agherzan, 2017-09-30:
Hi @fgiancane8 . Something looks fishy here. Why would we have egl and libegl. Aren't they the same thing? Maybe this was renamed in oe-core?

fgiancane8, 2017-09-30:
Hi Andrei,

The very same issue is bugging me. If you search the providers, we are
talking about the package virtual/egl. If you search in the recipe
xorg-xserver, it says that its PACKAGECONFIG[glamor] depends on libegl;
also check the mesa.inc file; it RPROVIDES libegl.

I had the same discussion on oe-core (which I reported in the link in my
first comment) but I was told that it is correct to have build time
dependencies on virtual/egl and runtime dependencies on libegl.

They said that it is correct that way and BSP graphics recipes should
RPROVIDE libegl...

Francesco Giancane

Via Paolo Braccini 46, 10141 Torino (TO)
+393933920466
https://it.linkedin.com/in/francesco-giancane-aa7213117

fgiancane8, 2017-09-30:
`PACKAGECONFIG[glamor] = "--enable-glamor,--disable-glamor,libepoxy,libegl"`

this is the unsatisfied dependency when choosing userland as PREFERRED_PROVIDER for openGL...

betafive, 2017-10-25:
The patch above does look correct based on the discussion on the oe-core mailing list. Could you submit this as a pull request?

fgiancane8, 2017-10-25:
Hi, thank you for the response.

Pull request submitted.

agherzan, 2017-10-30:
@fgiancane8 is can we close this issue now?

fgiancane8, 2017-10-30:
@agherzan OK to close. Thanks!

closed 2017-10-30 by agherzan