agherzan/meta-raspberrypi #142: Expansion error

naveen230, 2017-10-20:
I'm trying to build ubuntu based rpi3 image and I'm getting the following error,

ERROR: ExpansionError during parsing /poky/meta-raspberrypi/recipes-devtools/python/rpi-gpio_0.6.3.bb
Traceback (most recent call last):
  File "/poky/bitbake/lib/bb/data_smart.py", line 386, in DataSmart.expandWithRefs(s=' ${PYPI_SRC_URI}', varname='SRC_URI'):
                 try:
    >                s = __expand_var_regexp__.sub(varparse.var_sub, s)
                     try:
  File "/poky/bitbake/lib/bb/data_smart.py", line 111, in VariableParse.var_sub(match=<_sre.SRE_Match object; span=(1, 16), match='${PYPI_SRC_URI}'>):
                 else:
    >                var = self.d.getVarFlag(key, "_content", True)
                 self.references.add(key)
  File "/poky/bitbake/lib/bb/data_smart.py", line 737, in DataSmart.getVarFlag(var='PYPI_SRC_URI', flag='_content', expand=True, noweakdefault=False, parsing=False):
                     cachename = var + "[" + flag + "]"
    >            value = self.expand(value, cachename)

  File "/poky/bitbake/lib/bb/data_smart.py", line 410, in DataSmart.expand(s='${@pypi_src_uri(d)}', varname='PYPI_SRC_URI'):
         def expand(self, s, varname = None):
    >        return self.expandWithRefs(s, varname).value

  File "/poky/bitbake/lib/bb/data_smart.py", line 400, in DataSmart.expandWithRefs(s='${@pypi_src_uri(d)}', varname='PYPI_SRC_URI'):
                 except Exception as exc:
    >                raise ExpansionError(varname, s, exc) from exc

bb.data_smart.ExpansionError: Failure expanding variable PYPI_SRC_URI, expression was ${@pypi_src_uri(d)} which triggered exception TypeError: getVar() missing 1 required positional argument: 'expand'

Summary: There was 1 ERROR message shown, returning a non-zero exit code.

agherzan, 2017-10-20:
What version of each layer are you using?

naveen230, 2017-10-23:
Thanks for your reply!

Here are the layers I'm using
poky/meta \
  /poky/meta-poky \
  /poky/meta-yocto \
  /poky/meta-yocto-bsp \
  /poky/meta-debian \
  /poky/meta-raspberrypi \
  /poky/meta-openembedded \
  /poky/meta-openembedded/meta-python \ 
  /poky/meta-openembedded/meta-oe \
  /poky/meta-openembedded/meta-multimedia \
  /poky/meta-openembedded/meta-networking \
  /poky/meta-openembedded/meta-networking \
  /poky/meta-openembedded/meta-gnome \
  /poky/meta-openembedded/meta-xfce \
  /poky/oe-meta-go \

All layers use Morty branch except meta-raspberrypi which uses master branch

naveen230, 2017-10-23:
I tried different branches of meta-raspberrypi and I'm getting the same expansion error but with different files
ExpansionError during parsing /poky/meta-raspberrypi/recipes-core/images/rpi-test-image.bb
Traceback (most recent call last):
  File "/poky/meta/classes/image.bbclass", line 446, in __anon_510__poky_meta_classes_image_bbclass(d=<bb.data_smart.DataSmart object at 0x7f54ec08e940>):
             image_cmd = localdata.getVar("IMAGE_CMD", True)
    >        vardeps.add('IMAGE_CMD_' + realt)
             if image_cmd:
  File "/poky/bitbake/lib/bb/data_smart.py", line 569, in DataSmart.getVar(var='IMAGE_CMD', expand=True, noweakdefault=False, parsing=False):
         def getVar(self, var, expand, noweakdefault=False, parsing=False):
    >        return self.getVarFlag(var, "_content", expand, noweakdefault, parsing)

  File "/poky/bitbake/lib/bb/data_smart.py", line 737, in DataSmart.getVarFlag(var='IMAGE_CMD', flag='_content', expand=True, noweakdefault=False, parsing=False):
                     cachename = var + "[" + flag + "]"
    >            value = self.expand(value, cachename)

agherzan, 2017-10-23:
You need to use the same branch for meta-raspberrypi - morty that is. There are changes that are incompatible with old yocto version which are part of the master branch.

closed 2017-10-23 by agherzan