agherzan/meta-raspberrypi #146: fetch form pypi.python.org failed:  ERROR 403: SSL is required.

FrankBau, 2017-10-28:
**Description**

http: URIs are used in the recipes below, but no longer accepted by server. Changing to https will fix this.

**Steps to reproduce the issue:**
using pyro branch of meta-raspberry
1. $ bitbake -c fetch rpi-gpio
2. $ bitbake -c fetch rpio

**Describe the results you received:**
server complaines: ERROR 403: SSL is required.

**Describe the results you expected:**
the source is fetched.

**Additional information you deem important (e.g. issue happens only occasionally):**
seems to be fixed in master by inheriting pypi
could you please add the missing s for pyro too

**Additional details (revisions used, host distro, etc.):**

agherzan, 2017-10-28:
So this issue is reproducible on pyro, right? Rocko is not affected. Could you confirm?

FrankBau, 2017-10-29:
Right. Both recipes fail for pyro. 

Don't yet use rocko, but made a quick test:
rpi-gpio_0.6.3.bb uses already  `inherit pypi` and no explicit URL --> works
rpio_0.10.0.bb is still using the http:// URL fails with same ERROR 403: SSL is required.

hth

Frank

betafive, 2017-11-27:
It should be possible to modify the recipe to use the pypi class from the meta-python layer. This will set SRC_URI for us from the package name.

Would you like to take a look at that and send a patch?

betafive, 2018-03-05:
This should now be fixed in master and rocko.

PR #203 (2018-03-05): rpio: Use pypi.bbclass
This should resolve recent issues with the non-https URL in SRC_URI. 
fixes #146

**- What I did**

**- How I did it**
closed 2018-03-05 by betafive