agherzan/meta-raspberrypi #147: userland recipe packages split

fgiancane8, 2017-10-30:
Hi @agherzan ,

As pointed out in the previous mailing lists, I am providing some more information about this topic.
Just to recap: my understanding is that the official 'userland' package provided from Broadcom does not only contain *GL* libraries, but also other libraries required for interfacing with the hardware.
Please find attached the output of the 'ls' command on the 'image' directory of the 'userland' package directly from Yocto. (There is one file for bin directory, one for lib and one for include).
[files.bin.txt](https://github.com/agherzan/meta-raspberrypi/files/1427804/files.bin.txt)
[files.include.txt](https://github.com/agherzan/meta-raspberrypi/files/1427805/files.include.txt)
[files.lib.txt](https://github.com/agherzan/meta-raspberrypi/files/1427806/files.lib.txt)

Given that, in principle, we should be able to choose independently whatever provider we think for the *GL* stack, these are some possible scenarios:

- Deploy *GL* stack using 'mesa': 'userland' is not compiled, no extra library is deployed.
- Deploy *GL* stack using 'userland' : 'mesa-gl' will provide libGL.so and EGL, GLESv2 is taken from Broadcom. 

The net result is that if I want to use said libraries (e.g. for interfacing with the camera) I am forced to use 'mesa-gl' + 'userland'. We should split the 'userland' into 'userland' + 'userland-egl' + 'userland-libgles2' in order to be able to combine GL stacks without forgetting of other user space applications.

I reworked the 'userland' recipe for package split, but I need some hints to finish it because the new provider for egl and libglesv2 would be userland-egl and userland-libgles2 and not userland.

Will push a branch as soon  as possible with my starting point; writing two different recipes is doable but I think that maybe we can instruct Yocto better instead of compiling the package twice.

betafive, 2018-02-26:
@fgiancane8, any progress on this?

fgiancane8, 2018-02-26:
Hi @betafive ,

Did not have the time to elaborate more on this problem. Any suggestion on how to rework this recipe?

Thanks

fgiancane8, 2018-02-26:
The previous recipe had dependency issues and even though the package split was right, in the end it did not succeed in making the situation better.

agherzan, 2018-03-01:
I know @kraj is working with userland so he might give you some insights. I would say that pushing a PR is a good idea anyway and we can iterate on it.

fgiancane8, 2018-03-02:
Hi,

thanks for the suggestion.
I will start working on the recipe in these days and you will get a PR as soon as things get done.

kraj, 2018-03-03:
@fgiancane8 I think its fine to package more granularly, where you keep  virtual/egl virtual/gl and virtual/gles1 virtual/gles2 which can then be chosen based on vc4graphics in machine features. 

However, I did not understand your usecase clearly, are you trying to use libegl.so from mesa and libgles2 from userland ?

fgiancane8, 2018-03-03:
Hi @kraj 
userland does not only provide the GL stack (GLESv2, EGL, ...) but also userspace tools for Pi hw (camera, GPU interface, ...).

If you deploy the GL stack with mesa, then you do not have the userspace binaries.

I foresee those scenarios:
1) deploy entirely with userland -> you have GL stack + binaries
2) deploy with mesa -> you have mesa stack and no binaries

I want to achieve the possibility to deploy userspace binaries no matter what GL stack provider selected.

Hope this clarifies more the point, if not please ask.
Last time all I did was to add userland-egl and userland-gles2 to packages in userland recipe.

kraj, 2018-03-03:
I see. Thats totally valid and easy to achieve usecase. So I encourage you to go ahead with
patches. Let me know if you need help

fgiancane8, 2018-03-18:
Hi @kraj 

I followed the suggestions you gave me, and it is also what I did before.
This, unfortunately, will not work.

Package split is OK, but the problem is the PROVIDES variable: we declared _userland_ package to PROVIDE both libgles2 and libegl, but in reality, are the sub packages that PROVIDE those; i.e. userland-gles2 PROVIDES virtual/libgles2 and userland-egl PROVIDES virtual/egl (and then they RPROVIDES respective libraries).

What I am thinking at the moment is to patch the CMakeLists.txt of userland source in order to add a switch to prevent the GL libraries from compile; after that, we can write a userland-nogl recipe to build just the user spaces.

What do you think? @agherzan ?

kraj, 2018-03-18:
If you need that level of granularity during build then there is no other way than to build separate recipes for each of those however I consider this law of diminishing returns

fgiancane8, 2018-03-19:
hi @kraj ,
thanks for the comment. I patched the CMakeLists.txt for userland source code but with no avail; excluding the GL stack from building make the whole compile phase fail.

That CMakeLists needs to be revised, but as you said this kind of granularity becomes too heavy.

I think we can leave it as: either use broadcom "custom" libraries for GL and Camera OR we use mesa + v4l using a "standard" Linux behavior.

aswild, 2018-09-15:
It's been a few months, but I found this issue while dealing with the same problem. For 64-bit builds userland was masked due to the preferred provider for the GL recipes.

I found that a possible solution is to add a [userland-nographics](https://github.com/aswild/meta-newbs/blob/master/recipes-graphics/userland/userland-nographics_git.bb) recipe which builds the same stuff as userland, but then deletes the gles/egl stuff at the end of do_install.

Switching between userland and userland-nographics is done by setting the preferred provider for virtual/userland (checking for vc4graphics in MACHINE_FEATURES).

This solution has the advantage that the userland code would ever be built only once rather than multiple userland builds for the same image. An inefficiency is that userland-nographics still compiles its GL stack, but it's only a couple extra seconds on my machine.

If anyone thinks this approach could be suitable upstream, I can prepare a patch/PR.

kraj, 2018-11-28:
This is now addressed in entirety in thud/master

closed 2018-11-28 by kraj