agherzan/meta-raspberrypi #148: hciattach times out sometimes on first boot for rpi0w

keeslinp, 2017-11-01:
**Description**
I'm using this layer with yocto to build an image for the rpi0-w, and when I first boot the image on the device sometimes it works flawlessly but other times it doesn't set up the bluetooth properly. If it doesn't work on the first boot then rebooting will usually fix it eventually, sometimes running hciattach manually fixes the problem as well. When I dig through the logs I find that the hciattach has timed out during boot. 

**Steps to reproduce the issue:**
1. `bitbake rpi-hwup-image`
2. `dd` image onto sd card
3. boot

**Describe the results you received:**
Sometimes Bluetooth isn't working. There are no devices under `hcitool dev` or `hciconfig`. Also `rfkill` doesn't list the bluetooth device.

**Describe the results you expected:**
For bluetooth to setup the first time.

**Additional information you deem important (e.g. issue happens only occasionally):**
here is the output of `systemctl status brcm43438`:
```
Active: [[0;1;31mfailed[[0m (Result: exit-code) since Mon 2017-10-30 18:21:54 UTC; 28s ago
Process: 162 ExecStart=/usr/bin/hciattach -n /dev/ttyAMA0 bcm43xx 921600 noflow -[[0;1;31m(code=exited, status=1/FAILURE)[[0m
Main PID: 162 (code=exited, status=1/FAILURE)

systemd[1]: Started Broadcom BCM43438 bluetooth HCI.
hciattach[162]: Initialization timed out.
hciattach[162]: bcm43xx_init
systemd[1]: [[0;1;39mbrcm43438.service: Main process exited, code=exited, status=1/FAILURE[[0m
systemd[1]: [[0;1;39mbrcm43438.service: Unit entered failed state.[[0m
systemd[1]: [[0;1;39mbrcm43438.service: Failed with result `exit-code`.[[0m
```
**Additional details (revisions used, host distro, etc.):**
`local.conf`
```PACKAGE_CLASSES ?= "package_rpm"
BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
CONF_VERSION = "1"
CORE_IMAGE_EXTRA_INSTALL += "openssh rsync sensorhub"
IMAGE_INSTALL_append += "nodejs nodejs-npm bluez5 networkmanager linux-firmware-bcm43430 config systemd"
IMAGE_INSTALL_append += " crda i2c-tools hostapd linux-firmware-bcm43430 linux-firmware-ralink linux-firmware-rtl8192ce linux-firmware-rtl8192cu linux-firmware-rtl8192su wireless-tools dhcp-client iw wpa-supplicant"
DISTRO_FEATURES_append += " bluez5 bluetooth wifi systemd"```

keeslinp, 2017-11-01:
I've done some more testing, it seems that `systemctl restart brcm43438` fixes the problem consistently without a reboot. My band-aid solution is going to be adding a script on startup that checks if bluetooth is broken and if it is it will run that to fix it. Hopefully, that works until we can find a permanent solution :).

agherzan, 2017-11-27:
That sounds like a hardware issue. Are you able to reproduce this on a raspbian?

keeslinp, 2017-11-27:
I never had this issue on raspbian. I also tested it on 5+ pi's. For what it's worth, once I got the reset script working I never had any more problems. The key is to make sure that nothing tries to use bluetooth before it has been set up properly, otherwise, you have to reboot before it'll work again.

agherzan, 2017-12-01:
If you stop the service and run it manually after everything boots up, does it always work? I'm wondering if there is a dependency that we are not aware of.

keeslinp, 2017-12-01:
Yeah, that fixes it fairly consistently. I'll do some tests to see if anything have changed over the last month, but here is the script that I've been using that has worked consistently:

``` #!/bin/sh
if [ ! -d "/sys/class/bluetooth/hci0" ]; then
        sleep 20
        if [ ! -d "/sys/class/bluetooth/hci0" ]; then
                systemctl restart brcm43438
                sleep 20
                systemctl restart start-sensorhub
        fi
fi
echo nameserver 8.8.8.8 > /etc/resolv.conf```

agherzan, 2017-12-04:
It's definitely something that we need to fix for stabolity.

keeslinp, 2017-12-04:
Also, I tried removing my fix and the problem returned.

sherifomran, 2017-12-15:
i have the same issue, tested with systemv and systemd, i can not get the device to get initialized.

sherifomran, 2017-12-16:
i tried to reproduce the colleagues error. First I am using rpi0w, in the image, there is no /etc/firmware folder, the folder is installed in lib/firmware/ second it does not have the bcm43430A1.hcd file. As far as i know, the file is needed to reflash the chip. When i created the folders manually in mkdir -p /etc/firmware/brcm/ and copied the files to these locations, i get intialization time out.  I then copied the file LICENSE.broadcom_bcm43xx from /lib/firmware to /etc/firmware and i ran hciattach again, i get patch not found .. which patch does it mean? may be the firmware file BCM43430A1.hcd ? thank you

agherzan, 2017-12-21:
Moving https://github.com/agherzan/meta-raspberrypi/issues/174 discussion here. As per @sherifomran investigation this sounds like an outdated firmware. Is anybody able to provide a patch with an updated version of it to master? @keeslinp are you able to test this as well?

PR #183 (2018-01-23): Updates

**- What I did**

Update linux 4.9 and bluetooth firmware.
Fixes #148
closed 2018-01-23 by agherzan