agherzan/meta-raspberrypi #150: SPI and I2C

mihaelaapetroaie, 2017-11-02:
Hello, 

I am trying to enable SPI and I2C on a Raspberry PI 3. I am using yocto morty and meta-raspberrypi morty. All my layers are morty.

In the local.conf I have added the lines 
$ENABLE_SPI_BUS="1"
$ENABLE_I2C="1"

When my image boots, I get no i2c or spi in /dev, but after doing modprobe i2c-dev and modprobe spidev, i get an i2c-2, but no spidev* whatsoever.

It also seems that my yocto built os image has no config.txt.

The image I am building is rpi-hwup-image, althogh I also tried rpi-basic-image with the same results.

The machine I am using to build my rpi linux distro is Ubuntu 16.04.3 LTS.

Thank you!

agherzan, 2017-11-02:
What do you use as final image? The sdcard raw image? Or wic?

mihaelaapetroaie, 2017-11-02:
hello, 

Thanks for your answer. I actually have a minimal os, and on top of that I have OSTree. added the two lines to enable SPI and I2C in local.conf, rebuilt and sent the difference using OSTree to my yocto minimal image. 

However, after I saw your comment, I did a dd of the resulting image on a spare SD card, and i get both spidev* and i2c-* (two each). 

Do you, by any chance know why this happens?

Thanks a lot!

betafive, 2018-02-26:
I'm going to close this then as that's not something we can fix in meta-raspberrypi. You'll probably need to look at how this is handled for other targets with multiple partitions.

closed 2018-02-26 by betafive

agherzan, 2018-02-26:
Good catch @betafive .