agherzan/meta-raspberrypi #151: wic error

naveen230, 2017-11-02:
Hello 
I'm using pyro branches of meta-raspberrypi meta-mender meta-openembedded poky to create basic rpi image and getting following error because of the rootfs size.Any suggestions to overcome this issue?
7-sdimg/
| INFO: Creating image(s)...
|
| WARNING: overhead-factor was specified, but size was not, so bitbake variables will be used for the size. In this case both IMAGE_OVERHEAD_FACTOR and --overhead-factor will be applied
| ERROR: Actual rootfs size (17273 kB) is larger than allowed size 16384 kB
|
| + bb_exit_handler
| + ret=1
| + echo WARNING: exit code 1 from a shell command.
| WARNING: exit code 1 from a shell command.
| + exit 1
| ERROR: Function failed: do_image_sdimg (log file is located at /root/poky/build/tmp/work/raspberrypi3-poky-linux-gnueabi/rpi-basic-image/1.0-r0/temp/log.do_image_sdimg.3276)
ERROR: Task (/root/poky/meta-raspberrypi/recipes-core/images/rpi-basic-image.bb:do_image_sdimg) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4697 tasks of which 4691 didn't need to be rerun and 1 failed.

betafive, 2018-02-26:
I'm not sure I understand this issue - you're saying it's a wic error but it doesn't look like you're using wic. If this is still an issue could you send a the full bitbake output so we can see what's going wrong.

agherzan, 2018-02-26:
@betafive this error is a rootfs size restriction error. I bet that somewhere @naveen230 is using a size rootfs size restriction configuration. Looks completely unrelated to the BSP but let's give the user a couple of days to reply.

Sdcxv, 2018-03-21:
I encountered a similar problem, here is the layers I had added.
```
$ bitbake-layers show-layers
NOTE: Starting bitbake server...
layer                 path                                      priority
==========================================================================
meta                  /home/sdcxv/poky/meta-mender/openembedded-core/meta  5
meta-oe               /home/sdcxv/poky/meta-mender/meta-openembedded/meta-oe  6
meta-python           /home/sdcxv/poky/meta-mender/meta-openembedded/meta-python  7
meta-multimedia       /home/sdcxv/poky/meta-mender/meta-openembedded/meta-multimedia  6
meta-networking       /home/sdcxv/poky/meta-mender/meta-openembedded/meta-networking  5
meta-mender-core      /home/sdcxv/poky/meta-mender/meta-mender-core  6
meta-mender-demo      /home/sdcxv/meta-mender/meta-mender-demo  6
meta-raspberrypi      /home/sdcxv/poky/meta-mender/meta-raspberrypi  9
meta-mender-raspberrypi  /home/sdcxv/poky/meta-mender/meta-mender-raspberrypi  10
```
And here is my log files
```
$ cat log.do_image_sdimg
DEBUG: Executing python function set_image_size
DEBUG: 100228.000000 = 100228 * 1.000000
DEBUG: 212992.000000 = max(100228.000000, 212992)[212992.000000] + 1
DEBUG: 212992.000000 = int(212992.000000)
DEBUG: 212992 = aligned(212992)
DEBUG: returning 212992
DEBUG: Python function set_image_size finished
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/prelink/prelink_git.bb:do_populate_sysroot', '/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/quilt/quilt-native_0.65.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', '/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', '/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-core/glibc/glibc_2.27.bb:do_populate_sysroot', '/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/gcc/gcc-runtime_7.3.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/opkg-utils/opkg-utils_0.3.6.bb:do_populate_sysroot', '/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/gcc/gcc-cross_7.3.bb:do_populate_sysroot', '/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-core/meta/wic-tools.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/mtools/mtools_4.0.18.bb:do_populate_sysroot', '/home/sdcxv/poky/meta-mender/meta-raspberrypi/recipes-bsp/bootfiles/bcm2835-bootfiles.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/fdisk/gptfdisk_1.0.3.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-extended/parted/parted_3.2.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/dosfstools/dosfstools_4.1.bb:do_populate_sysroot', '/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/mklibs/mklibs-native_0.1.43.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-core/update-rc.d/update-rc.d_0.7.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/opkg/opkg_0.3.6.bb:do_populate_sysroot', '/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot', 'virtual:native:/home/sdcxv/poky/meta-mender/openembedded-core/meta/recipes-devtools/rsync/rsync_3.1.3.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['prelink-native', 'quilt-native', 'makedevs-native', 'ldconfig-native', 'glibc', 'gcc-runtime', 'opkg-utils-native', 'gcc-cross-arm', 'wic-tools', 'mtools-native', 'bcm2835-bootfiles', 'gptfdisk-native', 'pseudo-native', 'depmodwrapper-cross', 'parted-native', 'dosfstools-native', 'mklibs-native', 'update-rc.d-native', 'opkg-native', 'qemuwrapper-cross', 'pigz-native', 'rsync-native', 'automake-native', 'gnu-config-native', 'elfutils-native', 'binutils-native', 'autoconf-native', 'libtool-native', 'xz-native', 'systemd-systemctl-native', 'qemu-native', 'shadow-native', 'linux-libc-headers', 'libgcc', 'binutils-cross-arm', 'gmp-native', 'libmpc-native', 'mpfr-native', 'zlib-native', 'flex-native', 'texinfo-dummy-native', 'debianutils-native', 'openssl-native', 'popt-native', 'ncurses-native', 'util-linux-native', 'gettext-minimal-native', 'readline-native', 'pkgconfig-native', 'kmod-native', 'libarchive-native', 'libsolv-native', 'attr-native', 'acl-native', 'bison-native', 'm4-native', 'dtc-native', 'glib-2.0-native', 'pixman-native', 'libsdl-native', 'alsa-lib-native', 'cryptodev-linux-native', 'makedepend-native', 'lzo-native', 'python3-native', 'gtk-doc-native', 'e2fsprogs-native', 'bzip2-native', 'ninja-native', 'expat-native', 'cmake-n [... cut ...]
WARNING: /home/sdcxv/poky/meta-mender/openembedded-core/build/tmp-glibc/work/raspberrypi3-oe-linux-gnueabi/core-image-full-cmdline/1.0-r0/temp/run.do_image_sdimg.16475:1 exit 1 from 'BUILDDIR="/home/sdcxv/poky/meta-mender/openembedded-core/build" wic create "$wks" --vars "/home/sdcxv/poky/meta-mender/openembedded-core/build/tmp-glibc/sysroots/raspberrypi3/imgdata/" -e "core-image-full-cmdline" -o "$wicout/"'
+ exit 1
ERROR: Function failed: do_image_sdimg (log file is located at /home/sdcxv/poky/meta-mender/openembedded-core/build/tmp-glibc/work/raspberrypi3-oe-linux-gnueabi/core-image-full-cmdline/1.0-r0/temp/log.do_image_sdimg.16475)
```

Sdcxv, 2018-03-21:
I had added the following line into `conf/local.conf`
```
RPI_USE_U_BOOT = "1"
MENDER_PARTITION_ALIGNMENT_KB = "4096"
MENDER_BOOT_PART_SIZE_MB = "40"
IMAGE_INSTALL_append = " kernel-image kernel-devicetree"
IMAGE_FSTYPES_remove += " rpi-sdimg"
```

And run `$ bitbake core-image-full-cmdline` again, the problem solved.
```
Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "centos-7"
TARGET_SYS           = "arm-oe-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 = "master:086308aa2a5e332de6f00ed397c4a55d132f158f"
meta-oe
meta-python
meta-multimedia
meta-networking      = "master:4598769a96980b756b3489ac71ed46a075cfabd7"
meta-mender-core
meta-mender-demo     = "rocko:cb4feead40e48c4ece1a041c995defecd0e7ee96"
meta-raspberrypi     = "master:f86f9f0013e11311c66fc640fbb43b399606ca2f"
meta-mender-raspberrypi = "rocko:cb4feead40e48c4ece1a041c995defecd0e7ee96"
......
NOTE: Tasks Summary: Attempted 3304 tasks of which 2909 didn't need to be rerun and all succeeded.

Summary: There were 8 WARNING messages shown.
```

agherzan, 2018-03-22:
@Sdcxv can you reproduce this issue with only meta-rpi, meta-oe and poky layers?

naveen230, 2018-03-23:
I don't remember what I did but I'm able to develop the image now.Maybe I was using layers with different branches or using meta-debian caused this error.

nucatus, 2021-02-04:
Actually, this is because mender is handling the `sd_img` creation and the default partition size is too small. Setting the `MENDER_BOOT_PART_SIZE_MB = "32"` would fix the issue.