agherzan/meta-raspberrypi #152: bitbake gstreamer1.0-plugins-bad failes in rocko (with framebuffer: no x11, no wayland)

FrankBau, 2017-11-07:
**Description**

Trying to bitbake an image with framebuffer as graphical backend (local.conf: DISTRO_FEATURES_remove = " x11 wayland") and gstreamer 1.0 failes:

```console
$ bitbake gstreamer1.0-plugins-bad  
ERROR: Nothing PROVIDES 'vulkan' (but /home/bauer/raspi/rocko/recipes/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.12.2.bb DEPENDS on or otherwise requires it)
vulkan was skipped: one of '['x11', 'wayland']' needs to be in DISTRO_FEATURES
ERROR: Required build target 'gstreamer1.0-plugins-bad' has no buildable providers.
Missing or unbuildable dependency chain was: ['gstreamer1.0-plugins-bad', 'vulkan']
```

**Additional information you deem important (e.g. issue happens only occasionally):**

workaround: remove vulkan from DISTRO_FEATURES in local.conf

After removing vulkan, the image builds fine and the gst plugins bad work fine with framebuffer:

    # gst-launch-1.0 videotestsrc ! glupload ! glfiltercube ! glimagesink

**Additional details (revisions used, host distro, etc.):**

rocko. Looks like the same will happen in master

Don't know how to fix that. Since rocko: vulkan is in POKY_DEFAULT_DISTRO_FEATURES but not available for raspi ?!

mirzak, 2017-11-27:
Well IMO this is not related to meta-raspberrypi.

That is just how poky works nowadays.

If you remove wayland from DISTRO_FEATURES you must also remove vulkan, since vulkan depends on x11 or wayland.

E.g (extending your configuration).

```
DISTRO_FEATURES_remove = " x11 wayland vulkan"
```

mirzak, 2017-11-27:
or `gstreamer1.0-plugins-bad` needs to be updated to only include `vulkan` if `x11` or `wayland` is present in DISTRO_FEATURES

FrankBau, 2017-11-29:
Yes, I can remove vulkan from DISTRO_FEATURES, see my original posting. But: Raspi HW does not support vulkan, right? Wouldn't it be better when there were some cross-check in the raspi layer avoiding mis-configuration? 
One could also argue, like mirzak did, that the error is in the package recipe. Should all packages test for an intersection of MACHINE_FEATURES and DISTRO_FEATURES? 
Maybe I should derive my own x11-+ wayland-less distro from poky?

agherzan, 2017-11-29:
The question you have to ask is if this is a raspberrypi specific issue. Would other boards hit the same problem? If the answer is yes, we need to raise this issue against the core.