agherzan/meta-raspberrypi #153: raspberrypi3-64 does not build on master

loysollivier, 2017-11-09:
The raspberrypi3-64 does not build. Error: no rule to make target zImage.

**Steps to reproduce the issue:**
1. Follow quick start instructions
2. Use MACHINE="raspberrypi3-64"
3. bitbake rpi-hwup-image

**Describe the results you received:**
| make[2]: *** No rule to make target 'zImage'.  Stop.

**Describe the results you expected:**
A successful build

**Additional information you deem important (e.g. issue happens only occasionally):**
IMO related to:
https://github.com/agherzan/meta-raspberrypi/commit/50fd319205d8de9996edddad5f1a2c3f2b9e2f15
If I revert this commit it builds.

**Additional details (revisions used, host distro, etc.):**

betafive, 2017-11-09:
Ah I see the issue here. arm64 platforms don't support `zImage` as a kernel image format. They do support `Image.gz` but I'm not sure if the raspberrypi firmware is happy to boot that.

I'll have time to resolve this in a few days unless Andrei wants to jump in first.

agherzan, 2017-11-09:
Ah! arm64 doesn't provide a decompressor. And .gz needs to be handled by bootloader which I doubt bootcode does. Actually I'm pretty sure it does not. So we will have to revert this back to Image for this machine. It is a simple change. @loysollivier Do you want to push a PR?

loysollivier, 2017-11-09:
Thanks for the answers.

Sure if you want, doing it right away :)

agherzan, 2017-11-09:
@loysollivier you're the man!

martingkelly, 2017-11-09:
I just noticed this issue as well; thanks! How soon do you expect until this reaches master?

agherzan, 2017-11-09:
Very soon @martingkelly .

closed 2017-11-10 by agherzan

pesicsasa, 2018-06-14:
Hello, 
When trying to make an image of raspberrypi3-64, the process fails with the exact same error. Any thoughts on how to fix it? 

I am running/building on linux ubuntu 17.10, x64. Following these instructions:https://resinos.io/docs/custombuild/. The command i use is ./resin-yocto-scripts/build/barys -m raspberrypi3-64 --resinio.

betafive, 2018-06-14:
@pesicsasa, I'd recommend raising that issue with Resin OS.

commit 4d3a148b (2017-11-10): raspberrypi3-64: set default kernel to Image

Fixes: 50fd319205d8 for raspberrypi3-64.
Fixes: #153

For raspberrypi3-64 set default kernel image to "Image".

"zImage" are not supported by arm64 platforms. And ".gz" images are not
handled by bootloader yet.